nxp-archive/openil_sja1105-tool #1: Link to the bitbake recipe readme is down

hoedur, 2017-06-06:
The link to the bitbake recipe readme provided in the README.md file is down:
http://sw-stash.freescale.net/users/b56933/repos/sja1105-tool-bitbake/browse

The domain sw-stash.freescale.net seems to be down for some time. However, I have not found any replacement or backup of this site (no, archive.org didn't help either).

vladimiroltean, 2017-06-07:
Hi,
Thanks a lot for your interest. May I ask what is your exact intention?
The link is to an internal server. It was not updated since the push to Github.
The OpenIL project has since moved from Bitbake to Buildroot. We do no longer use the Bitbake recipe.
-Vladimir

hoedur, 2017-06-23:
Hi,
I was curious about the sja1105 tool, it provides a pretty cool configuration interface for this switch. I am currently using it in a prototype.
I do not really need a bitbake recipe, but since this is a public repository, I wanted to report a possible mistake in the documentation ;)

vladimiroltean, 2017-07-27:
Fixed with this commit: 3e155a8693f015a28793d6d9e582c10b1cc31801
Closing.

closed 2017-07-27 by vladimiroltean