nxp-archive/openil_sja1105-tool #2: Possible issue with the xMII mode parameters table

610blixten, 2017-06-30:
Hello,

I'm using the sja1105-tool to configure a couple of switches in a prototype design that I'm working with. Currently I'm debugging a bad ethernet TX signal.

One thing that I have encountered is that in the sja1105-tool XML configuration, in the xMII mode parameters entry, I have to set the phy_mac value to 0 (PHY mode) for the ports to work (badly). But the port should actually be configured as a MAC since it is connected directly to a phy via RMII.

Is this something that you recognize or do I have faulty RMII connection? 

Can I utilize the sja1105-tool in some way, to easily read the parameters that are stored in and used by the switch?

Best regards,

vladimiroltean, 2017-07-03:
Hi,
I could only test the sja1105-tool over RGMII interfaces so far (this is what the LS1021ATSN board has). The existing code for RMII is incomplete and not expected to work (The PLLs are not configured properly). I am working on a fix.
Thanks,
Vladimir

vladimiroltean, 2017-07-03:
* In RMII, the 50MHz REF_CLK is always an input to the PHY. It can either be driven by the MAC or by an external oscillator, case in which it's also an input to the MAC.
* If I understand correctly, you don't have an external oscillator and are relying on the SJA1105 switch to drive the REF_CLK. If this is true, PHY_MAC should be left as XMII_MODE_MAC (0) in order to have the Clock Generation Unit drive the REFCLK from its internal PLL1 (50MHz).
* The issue here is that PLL1 was not configured, meaning that REFCLK was probably spilling out junk instead of 50MHz.
* Changing PHY_MAC to XMII_MODE_PHY was not the way to go, since now REFCLK is probably tuned to noise (nobody is driving it).
* There now exists a new commit on the [feature/rmii](https://github.com/openil/sja1105-tool/tree/feature/rmii) branch, that does its best to configure PLL1 according to the UM10944 user manual. Can you please test it and report back?

610blixten, 2017-07-13:
Hi!

Thanks for your input and sorry for late reply! I've been swamped over here.

Actually, in our case, the REF_CLK is an output from the PHY (KSZ8061RNB). The REF_CLK should be an input to the MAC. 

So with the old source code, if I configure the port as a PHY it will be driven by the external PHY and it works partly. Only some devices are able to interpret the generated TX signal. I'm not sure what's wrong with the signal and I'm not sure about all the implications of connecting a PHY to a PHY over RMII.

If I configure the port as a MAC, it will generate a 25 MHz REF_CLK and nothing works. Naturally.

I will check out your patch and dive into the CGU documentation. Hopefully I'll be able to get this thing running before the weekend! I'll let you know how things go. ;-)

vladimiroltean, 2017-07-14:
I **think** what you are trying to do is unsupported. Let me explain.

According to the [RMII spec](http://ebook.pldworld.com/_eBook/-Telecommunications,Networks-/TCPIP/RMII/rmii_rev12.pdf), the MAC drives TXD[1:0] while the PHY drives RXD[1:0].

Now, the PHY can transmit extra "out-of-band signals" via RXD[1:0].
These "out-of-band signals" are practically some extra symbols sent prior to the preamble of each frame. Please refer to **FIGURE 2. 100 Mb/s Reception with no errors**: there are some groups of symbols on RXD[1:0] called /J/ and /K/.

In the RMII spec, the MAC does not have this out-of-band signaling mechanism. You can see this is the case because in **FIGURE 4. 100 Mb/s Transmission**, TXD[1:0] begins right away with the preamble, as soon as TX_EN is asserted.

Up till now, everything was purely theory. Hang on for more.

There is a shortcoming in the SJA1105T Clock Generation Unit. It cannot operate in RMII-MAC mode with an external 50MHz clock source. Let's say it just adheres a little "too closely" to the spec, which is super loose, but bottom line is, if you generate the REF_CLK externally to the SJA1105T, you must configure it as RMII-PHY.

Otherwise, if you configure it as RMII-MAC, it is simply not configurable enough, and will also attempt to drive the REF_CLK itself.

Now comes the issue you're probably having. Setting the SJA1105T in RMII-PHY mode has some unintended and unwanted consequences. The SJA1105T emulates a PHY interface fully. This means that it will include some extra out-of-band signals (those /J/ and /K/) on the RXD[1:0] prior to the start of some frames.

Now your PHY chip (which I haven't looked at very closely) will probably not understand those extra /J/ /K/ symbols on RXD[1:0], since theoretically it's not required to - a PHY-to-PHY connection is not supposed to happen. What it will probably do is simply copy and modulate those extra symbols (along with the rest of the frame) onto the 100Base-TX wire.

This is probably the reason why communication with some devices connected to the RNB PHY will work and other devices don't. Those that don't simply expect a valid preamble on the other side of RXD[1:0], when what they really get are some extra symbols generated by the SJA1105T functioning in PHY mode. The whole frames containing such symbols will get discarded.

Bottom line: All of the above is speculation, which may or may not have roots in reality. You will have to check if the KSZ8061RNB is compatible with a PHY-to-PHY connection, or if you can disable and re-route its REF_CLK, and leave the SJA1105T in RMII-MAC mode. But one thing's for sure: my patch won't help you in the situation you're in right now :)

610blixten, 2017-07-14:
Great answer! Thanks!

I kind of figured out that this was not solvable in software. Now it's an issue for the hardware guys.

Have a nice weekend! :-)

closed 2017-07-14 by 610blixten