nxp-archive/openil_sja1105-tool #8: Inconsistent return codes

hoedur, 2017-10-02:
I observed that the sja1105-tool does return nonzero values even if execution was successful. The easiest way to reproduce is by issuing a simple command such as
`sja1105-cool reset cold`
This command returns '8' upon success if executed on an arm64 machine. As a side effect, the sja1105-tool does not even print "ok" at the end of its execution - it only does `if (rc == 0)`. This makes it really hard to use the tool in scripts or otherwise automated environments like systemd units.

vladimiroltean, 2017-10-03:
Hi Po,

I'd like not to change the `if (rc == 0)` check.
The call path is like this:

- `src/lib/reset/rgu.c`: `sja1105_reset()` returns `sja1105_spi_transfer()`
- `src/lib/spi/spi-transfer.c`: `sja1105_spi_transfer()` returns `ioctl(..., SPI_IOC_MESSAGE(1), ...)`;

In man ioctl(2):

> 
> RETURN VALUE
>        Usually, on success zero is returned.  A few ioctl() requests use the return value as an output parameter and return a nonnegative value on success.  On error, -1 is returned,  and  errno  is
>        set appropriately.

It looks like the spidev ioctl is one of those returning the total length of transfers on success, and not zero: https://github.com/openil/linux/blob/master/drivers/spi/spidev.c#L250

- `drivers/spi/spidev.c`: `spidev_ioctl()` returns `spidev_message()`
- `drivers/spi/spidev.c`: `spidev_message()` returns `total` which is the number of transferred bytes.

So I believe the solution here would be in `sja1105_spi_transfer()` to just `return (rc == size) ? 0 : -1;`. I hope that won't create further problems.

closed 2017-12-18 by vladimiroltean

vladimiroltean, 2018-05-18:
As of current master branch HEAD, the sja1105-tool returns the following error codes to other calling processes:
https://github.com/openil/sja1105-tool/blob/master/src/lib/include/errors.h