nxp-archive/openil_sja1105-tool #20: Standard.xml file

echizenr, 2018-08-14:
Hello,

I'm currently trying to load the standard.xml file with the following command : **sja1105-tool config load standard.xml** and I always have this result : 

> Retagging Table unimplemented!
> Token "sgmii-table" matches no known command. Try : 
> * schedule-table
> *schedule-entry-points-table
> *vl-lookup-table
> *vl-policing-table
> *vl-forwarding-table
> *l2-address-lookup-table
> *l2-policing-table
> * vlan-lookup-table
> * l2-forwarding-table
> * mac-configuration-table
> * schedule-parameters-table
> * schedule-entry-points-parameters-table
> *vl-forwarding-parameters-table
> *l2-address-lookup-parameters-table
> * l2-forwarding-parameters-table
> *clock-synchronisation-parameters-table
> *avb-parameters_table
> *general-parameters-table
> *retagging-table
> *xmii-mode-parameters-table
> parse_config_table_failed
> Could not parse static config from XML !

Could you help me please ?

vladimiroltean, 2018-08-14:
```
Retagging Table unimplemented!
```
This is a warning and can be ignored.
```
Token "sgmii-table" matches no known command
```
The `sgmii-table` is a recent addition (https://github.com/openil/sja1105-tool/commit/415ecfeefb441db1c314892877592b90a796df67) for P/Q/R/S chips.
What the error means is that your `sja1105-tool` is older (you can check with `sja1105-tool --version`) and doesn't understand the `<sgmii-table/>` line in the xml.
What you can do:
* Generate the xml with the same `sja1105-tool` version:
```
sja1105-tool config default ls
sja1105-tool config save standard.xml
sja1105-tool config load standard.xml
```
* Remove the `sgmii-table` from your existing xml
* Update the `sja1105-tool`.

By the way, while it's unfortunate that old versions have become xml-incompatible with new ones, I expect that to no longer be the case in the future, as all new tables of the new P/Q/R/S generation of chips are now in place.

echizenr, 2018-08-14:
Ok I understand now. That's weird because I clone the most recent version on github.

Thanks a lot, It's working now.

I close this issue.

vladimiroltean, 2018-08-14:
Generally this happens when you generate the xml with a newer `sja1105-tool` than the one you use to load it.
Maybe you generated the xml on the PC and loaded it on the board, which had an old tool?

echizenr, 2018-08-14:
Yes I generated the xml on the PC and load it on the board with scp.

I have a problem with my web interface. When I start the traffic, this message appear : 

[image]

Do you know how can I solve it ?

vladimiroltean, 2018-08-14:
The node at IP 172.15.0.3 is unreachable. Adapt the address to the machine you want to connect to.

echizenr, 2018-08-14:
But according to the user guide, my eth2 (Board 3) is configured with the IP address 172.15.0.3, eth0 with 192.168.15.3 and my host PC in the managing network is configured with the IP address 198.162.15.100.

I'm suppose to change the IP addresse of eth2 but if I do that it won't be able to reach Board 1 & 2 eth2 and eth3.

vladimiroltean, 2018-08-14:
Does the machine from which you run the web application have any Ethernet interface in the 172.15.0.0/24 network? Are you able to ping between them? If the route involves SJA1105 switches, are they initialized? What does `sja1105-tool status ports` show?

echizenr, 2018-08-14:
I have only 1 Ethernet interface on the machine which run the web application and it is configure in the 192.168.15.0/24 network.

All boards are able to ping the web application machine.

echizenr, 2018-08-14:
I don't know why but it's working now, maybe IP address wrong when I set them during uboot.

Thanks for your help !

vladimiroltean, 2018-08-14:
If you are starting the web server by hand (`/usr/lib/node_modules/lbt/server.js`), does the output give any cues?
If you start the web server through the `/etc/init.d/S95lbt` init script, then the log is available in the `/var/log/lbt.log` file.

> It appear when I'm trying to run both Iperf flows together
I suspect you are trying to open 2 iperf servers on the same port.

vladimiroltean, 2018-08-16:
Change the second iperf port from 5201 to something else, like 5202. You cannot have multiple processes listening on the same TCP port on the same machine.

echizenr, 2018-08-16:
Ok, it seams that it work now.

Thank you, I will be able to perform the other tests.

I close this issue.

closed 2018-08-16 by echizenr