nxp-archive/openil_sja1105-tool #21: sja1105-tool prioritizing.xml

echizenr, 2018-08-16:
Hello,

I think my prioritizing.xml file isn't working because on the web application output, I get the same result as without flow priorities. (I'm trying to run both iPerf flow at the same time in order to see something like in figure 39 (user guide 7.9.6.2).

You can find bellow the output from the LBT : 

[image]

Output from board 1 which run the lbt server : 

> 
> PUT /running
> traffic start/stop: old state false, new state true
> sse :: established new connection to ::ffff:192.168.15.100
> iPerf (Board 2 -> Board 3) iperf Destination :: conn ready
> iPerf (Board 1 -> Board 3) iperf Destination :: conn ready
> iPerf (Board 2 -> Board 3) iperf Destination STDOUT: -----------------------------------------------------------
> iPerf (Board 2 -> Board 3) iperf Destination STDOUT: -----------------------------------------------------------
> iPerf (Board 1 -> Board 3) iperf Destination STDOUT: -----------------------------------------------------------
> iPerf (Board 1 -> Board 3) iperf Destination STDOUT: -----------------------------------------------------------
> iPerf (Board 1 -> Board 3) iperf Client :: conn ready
> iPerf (Board 2 -> Board 3) iperf Client :: conn ready
> iPerf (Board 1 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 47768
> iPerf (Board 1 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5201 connected to 172.15.0.1 port 47770
> iPerf (Board 2 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 47172
> iPerf (Board 2 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5202 connected to 172.15.0.2 port 47174
> iPerf (Board 1 -> Board 3) iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> iPerf (Board 2 -> Board 3) iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> 

I run followings command on board 2 : 

> sja1105-tool config default ls
> sja1105-tool config save prioritizing.xml
> sja1105-tool config load prioritizing.xml
> sja1105-tool config upload

Do you have an idea how can I solve my problem ?

vladimiroltean, 2018-08-16:
Yes. A `prioritizing.xml` script generated using your commands above is in fact prioritizing just in the name (therefore, default, non-prioritizing behavior).
The prioritizing configuration can be obtained by running [this script](https://github.com/openil/sja1105-tool/blob/master/src/helpers/configs/rate-limiting/prioritizing.sh). More precisely, these lines are what assign higher VLAN priority to traffic coming from one port, compared to another:
```
# Flow 1 is coming from Board 1. From Board 2's perspective, this
# is connected to chassis port ETH3 (RGMII 2).
# Flow 2 is coming from Board 2. This is the internal port RGMII4.
sja1105-tool config modify mac-config[2] vlanprio ${flow1_prio}
sja1105-tool config modify mac-config[4] vlanprio ${flow2_prio}
```

vladimiroltean, 2018-08-16:
You said right above that you generate the xml as follows:
```
sja1105-tool config default ls
sja1105-tool config save prioritizing.xml
sja1105-tool config load prioritizing.xml
sja1105-tool config upload
```
Step by step:
* `sja1105-tool config default ls`: Generate the built-in default configuration for the `ls1021atsn` board (shortened as `ls`), and save it to the binary file `/lib/firmware/sja1105.bin` (also called "staging area").
* `sja1105-tool config save prioritizing.xml`: Reads the staging area (generated previously) and exports it in xml format to the `prioritizing.xml` file.
* `sja1105-tool config load prioritizing.xml`: Reads the `prioritizing.xml` file and exports it to the staging area (`/lib/firmware/sja1105.bin`). Yes, this is redundant.
* `sja1105-tool config upload`: Reads the staging area and sends it over SPI to the hardware.

So as you can see, the command sequence above overrides any `prioritizing.xml` file you might have.
A correct sequence would look like:
```
[on the pc, in the helpers folder]
source envsetup
./configs/rate-limiting/prioritizing.sh --flow1-prio 1 --flow2-prio 2
scp ./configs/rate-limiting/prioritizing.xml root@192.168.15.2:.
[on the board at 192.168.15.2]
sja1105-tool config load -f prioritizing.xml
```

echizenr, 2018-08-16:
No, I generate the xml file with : 

> source envsetup
> ./prioritizing.sh --flow1-prio 5 --flow2-prio 3

And I use the following commands for loadding the configuration : 

> sja1105-tool config default ls
> sja1105-tool config save prioritizing.xml
> sja1105-tool config load prioritizing.xml
> sja1105-tool config upload

 Sorry for the misunderstanding !

I tried this in order to be in the same situation you explained me on the previous issue named "standard.xml file" (the problem with sja1105-tool version).

So I remove the sgmii-table from my xml file and it's working now

Thank you for your help !

closed 2018-08-16 by echizenr

vladimiroltean, 2018-08-16:
If you update the sja1105-tool on your board as well, you won't even need to remove the sgmii-table from the xml.

echizenr, 2018-08-16:
Yes, I will do that for the next test.

Thanks again ! you really helped me.