nxp-archive/openil_sja1105-tool #22: Error directory xml generation

echizenr, 2018-09-04:
Hello,

I have a problem with sja1105-tool. I'm trying to generate xml file for TSN demo and when I run this command : 

> for i in 1 2 3; do ./configs/synchronized-qbv/qbv-ptp.sh --board ${i}; scp ./configs/
synchronized-qbv/qbv-ptp-board${i}.xml root@192.168.15.${i}:.; done

I obtain the following comment : 

> Line 10 : /home/sja1105-tool-1.0/src/helpers/_build/usr/bin/sja1105-tool: No such file or directory

I already run the source envsetup command and chmod.

I also did the Prerequisites steps (Linaro and libxml2-2.9.3).

Could you help me ?

I thank you in advance

vladimiroltean, 2018-09-04:
The `sja1105-tool: No such file or directory` error is indicative of the fact that you are trying to run a binary compiled for a different architecture. `file sja1105-tool` should clarify. To run `sja1105-tool` on PC you need to compile for x86_64, not use the ARMv7 cross-compiler.

vladimiroltean, 2018-09-04:
From the folder you're in (`/home/sja1105-tool-1.0/src/helpers`):
```
make -C ../../ clean
rm -rf _build
./setup.sh
```

echizenr, 2018-09-04:
Thank you !
It seems I had another mistake, I didn't run the setup.sh scripts.

It's all working now

I close this issue

closed 2018-09-04 by echizenr