nxp-archive/openil_sja1105-tool #23: LBT error with prioritizing.xml and policing.xml

echizenr, 2018-09-18:
Hello,

I've got a problem with my LBT web application. When I load the prioritizing.xml conf and run the test, Iperf1 flow work normally but Iperf2 flow (192.168.15.2:22 to 172.15.0.3:22) stops the LBT application with the next return : 

> Command "iperf3 -t 86400 -p 5202  -c 172.15.0.3" on root@192.168.15.2:22 exited with code undefind
> GET /flows
> GET /running
> Command "iperf3 -1 -f m -i 0.5 -s -p 5202" on root@172.15.0.3:22 exited with code undefined and sd
> PUT /flows
> {"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},"destination":{"}
> Successfully written flows to file.
> Command "iperf3 -1 -f m -i 0.5 -s -p 5202" on root@172.15.0.3:22 exited with code undefined and sd
> /usr/lib/node_modules/lbt/server.js:450
>             state.plotter[flowType].stdin.end();
>                                          ^
> 
> TypeError: Cannot read property 'end' of undefined
>     at forEach (/usr/lib/node_modules/lbt/server.js:450:42)
>     at Array.forEach (native)
>     at stopTraffic (/usr/lib/node_modules/lbt/server.js:443:23)
>     at Channel.stream.on (/usr/lib/node_modules/lbt/server.js:230:17)
>     at emitOne (events.js:101:20)
>     at Channel.emit (events.js:188:7)
>     at Channel.<anonymous> (/usr/lib/node_modules/lbt/node_modules/ssh2/lib/Channel.js:135:18)
>     at Channel.g (events.js:292:16)
>     at emitNone (events.js:91:20)
>     at Channel.emit (events.js:185:7)

Strange fact, when I run both flows together, it work normally. The same thing happen when I load the policing.xml configuration.

Could you help me ?
I thank you in advance.

vladimiroltean, 2018-09-18:
I can explain why this happens, but I'm afraid it's a limitation in the way the LBT is currently implemented.
The exception 

> TypeError: Cannot read property 'end' of undefined 

occurs [here](https://github.com/openil/openil/blob/master/package/nodejs-lbt/src/server.js#L450) (in the stopTraffic function). So you already were on a error condition, it's just that the exception is making it worse.

What you can try to do is apply the patch below, which avoids the second error, and repeat the test:

```
From c9770422b81ac1c3a4565ce00273e425c65b2c45 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Tue, 18 Sep 2018 18:43:25 +0300
Subject: [PATCH] nodejs-lbt: server: avoid exception when stopTraffic called
 and plotter not yet started

---
 package/nodejs-lbt/src/server.js | 4 +++-
 1 file changed, 3 insertions(+), 1 deletion(-)

diff --git a/package/nodejs-lbt/src/server.js b/package/nodejs-lbt/src/server.js
index ddcbed5..ef31697 100755
--- a/package/nodejs-lbt/src/server.js
+++ b/package/nodejs-lbt/src/server.js
@@ -447,7 +447,9 @@ function stopTraffic(error) {
                 if (typeof(f.srcSSHConn) != "undefined") { f.srcSSHConn.end() };
                 if (typeof(f.dstSSHConn) != "undefined") { f.dstSSHConn.end() };
             });
-            state.plotter[flowType].stdin.end();
+            if (typeof(state.plotter[flowType].stdin) != "undefined") {
+                state.plotter[flowType].stdin.end();
+            };
         }
     });
     state.clients.forEach((stream) => {
-- 
2.7.4
```

But the original errors that occured are these:
*  [in the onDestinationSSHConnReady function](https://github.com/openil/openil/blob/master/package/nodejs-lbt/src/server.js#L222).
* [in the onSourceSSHConnReady function](https://github.com/openil/openil/blob/master/package/nodejs-lbt/src/server.js#L142)

Basically what fails is the SSH connection between nodes 2 and 3 (the link restricted by the policer).
Normally what should happen is that LBT should try to establish SSH connections through the management interfaces (192.168.15.x) and send iperf through the data interfaces (172.15.0.x), but currently, [the IP of the SSH destination is the same as the IP of the iperf destination](https://github.com/openil/openil/blob/master/package/nodejs-lbt/src/server.js#L384-L389), which of course makes both iperf and SSH traffic get routed onto the same interface.

Does this policing.xml that you're using contain ingress policer values that are documented in the user manual, or did you alter the bandwidth limits manually to lower values? During my testing this scenario hasn't happened with what was documented.

echizenr, 2018-09-19:
It is weird because when I ran those tests few weeks ago, it work normally. But now, it doesn't work.

I used value specified in the user manual in order to have a normal behavior before trying to change bandwidth limitation and vlan priorities.

how can I apply the patch ? Did I need to modify the server.js file ?

vladimiroltean, 2018-09-19:
> how can I apply this patch ?

just edit the /usr/lib/node_modules/lbt/server.js on board, go to line 447 and apply the changes there.

echizenr, 2018-09-19:
I applied the patch but it's still not working. 
Did I need to use the OpenIL master branch if I use sja1105-tool master branch ?

vladimiroltean, 2018-09-19:
> I applied the patch but it's still not working.

Did you read my explanation on what this patch does, and what is the other limitation in LBT that leads to SSH connections being cut off by the policer?

vladimiroltean, 2018-09-19:
The log that you attached in the first post is truncated. Could you please attach the full version?

echizenr, 2018-09-19:
Sure, here are the full version : 

> [root@OpenIL:~]# /usr/lib/node_modules/lbt/server.js &
> [root@OpenIL:~]# {"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},
> "destination":{"user":"root","hostname":"172.15.0.3","port":22},"port":5201,"transport":"tcp","ban
> dwidth":-1,"enabled":true,"label":"Host 1","id":"c86cd7a0-d7da-400a-829c-a8b3f8e9dca4"},{"source":
> {"user":"root","hostname":"192.168.15.2","port":22},"destination":{"user":"root","hostname":"172.1
> 5.0.3","port":22},"port":5202,"transport":"tcp","bandwidth":-1,"enabled":true,"label":"Host 2","id
> ":"8c1309e7-65a4-415d-9b7a-d7cc8a74accf"}],"ping":[]}}
> Server listening for http requests on port 8000
> GET /
> GET /client.js
> GET /flows
> GET /running
> PUT /flows
> {"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},"destination":{"u
> ser":"root","hostname":"172.15.0.3","port":22},"port":5201,"transport":"tcp","bandwidth":-1,"enabl
> ed":true,"label":"Host 1","id":"142178ca-ef88-4547-81b1-265595a40e98"},{"source":{"user":"root","h
> ostname":"192.168.15.2","port":22},"destination":{"user":"root","hostname":"172.15.0.3","port":22}
> ,"port":5202,"transport":"tcp","bandwidth":-1,"enabled":false,"label":"Host 2","id":"baeece05-be9d
> -4270-825d-acbcbe2161f2"}],"ping":[]}}
> Successfully written flows to file.
> PUT /running
> traffic start/stop: old state false, new state true
> sse :: established new connection to ::ffff:192.168.15.100
> Host 1 iperf Destination :: conn ready
> Host 1 iperf Destination STDOUT: -----------------------------------------------------------
> Host 1 iperf Destination STDOUT: -----------------------------------------------------------
> Host 1 iperf Client :: conn ready
> Host 1 iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 37596
> Host 1 iperf Destination STDOUT: [  5] local 172.15.0.3 port 5201 connected to 172.15.0.1 port 375
> 98
> [   65.223156] random: crng init done
> Host 1 iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> PUT /running
> traffic start/stop: old state true, new state false
> GET /running
> GET /flows
> feedgnuplot process exited with code 0
> Command "iperf3 -1 -f m -i 0.5 -s -p 5201" on root@172.15.0.3:22 exited with code undefined and si
> gnal undefined
> Command "iperf3 -t 86400 -p 5201  -c 172.15.0.3" on root@192.168.15.1:22 exited with code undefine
> d and signal undefined
> GET /flows
> GET /running
> PUT /flows
> {"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},"destination":{"u
> ser":"root","hostname":"172.15.0.3","port":22},"port":5201,"transport":"tcp","bandwidth":-1,"enabl
> ed":false,"label":"Host 1","id":"3c2bcb9a-1cfa-4e9a-9acb-dde19f6bcb72"},{"source":{"user":"root","
> hostname":"192.168.15.2","port":22},"destination":{"user":"root","hostname":"172.15.0.3","port":22
> },"port":5202,"transport":"tcp","bandwidth":-1,"enabled":true,"label":"Host 2","id":"11cf960c-6aa4
> -4235-9232-f47fb6f18881"}],"ping":[]}}
> Successfully written flows to file.
> PUT /running
> traffic start/stop: old state false, new state true
> sse :: established new connection to ::ffff:192.168.15.100
> Host 2 iperf Destination :: conn ready
> Host 2 iperf Destination STDOUT: -----------------------------------------------------------
> Host 2 iperf Destination STDOUT: -----------------------------------------------------------
> Host 2 iperf Client :: conn ready
> Host 2 iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 37382
> Host 2 iperf Destination STDOUT: [  5] local 172.15.0.3 port 5202 connected to 172.15.0.2 port 373
> 84
> Host 2 iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> PUT /running
> traffic start/stop: old state true, new state false
> Command "iperf3 -t 86400 -p 5202  -c 172.15.0.3" on root@192.168.15.2:22 exited with code undefine
> d and signal undefined
> GET /flows
> GET /running
[... 22 lines cut ...]
> Host 2 iperf Client :: conn ready
> Host 1 iperf Client :: conn ready
> Host 1 iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 37612
> Host 2 iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 37386
> Host 1 iperf Destination STDOUT: [  5] local 172.15.0.3 port 5201 connected to 172.15.0.1 port 376
> 14
> Host 2 iperf Destination STDOUT: [  5] local 172.15.0.3 port 5202 connected to 172.15.0.2 port 373
> 88
> Host 1 iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> Host 2 iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> PUT /running
> traffic start/stop: old state true, new state false
> feedgnuplot process exited with code 0
> Command "iperf3 -t 86400 -p 5202  -c 172.15.0.3" on root@192.168.15.2:22 exited with code undefine
> d and signal undefined
> Command "iperf3 -1 -f m -i 0.5 -s -p 5202" on root@172.15.0.3:22 exited with code undefined and si
> gnal undefined
> Command "iperf3 -1 -f m -i 0.5 -s -p 5201" on root@172.15.0.3:22 exited with code undefined and si
> gnal undefined
> Command "iperf3 -t 86400 -p 5201  -c 172.15.0.3" on root@192.168.15.1:22 exited with code undefine
> d and signal undefined
> GET /flows
> GET /running
> GET /flows
> GET /running

vladimiroltean, 2018-09-19:
> Host 1 iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 37612
> Host 2 iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 37386
> (...)
> PUT /running
> traffic start/stop: old state true, new state false

`PUT /running` means that somebody is telling the server to stop traffic (`old state true, new state false`).
There's no error condition in this log, it's just the client who's requesting the traffic stop. The only place where `PUT /running` is called from the client is from `btnStartStop.onclick`.

echizenr, 2018-09-19:
Yes I did stop the traffic but it was because no graph output were shown by LBT : when I start the Iperf 2 flow, the LBT is stuck on the main page.

vladimiroltean, 2018-09-19:
> when I start the Iperf 2 flow, the LBT is stuck on the main page

So no error, but no plotting either?
Can you please apply this patch? It allows you to see the whole output of the iperf processes. I want to see if iperf still starts (since no errors are obvious from the log above). Perhaps the iperf output format is different for some reason in the failing case, and that's why no plotter data is sent to the browser.

```
From b05546e536c377a602c6667842428fab312c87a1 Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Wed, 19 Sep 2018 15:37:27 +0300
Subject: [PATCH] nodejs-lbt: enable logging of iperf commands stdout

---
 package/nodejs-lbt/src/server.js | 9 ++++-----
 1 file changed, 4 insertions(+), 5 deletions(-)

diff --git a/package/nodejs-lbt/src/server.js b/package/nodejs-lbt/src/server.js
index ef31697..c571671 100755
--- a/package/nodejs-lbt/src/server.js
+++ b/package/nodejs-lbt/src/server.js
@@ -172,8 +172,8 @@ function onSourceSSHConnReady(flowType) {
                 }
             } else {
                 /* If not taking reports, just print out the output as-is. */
-                //console.log("%s %s Source :: STDOUT: %s",
-                        //this.label, flowType, line);
+                console.log("%s %s Source :: STDOUT: %s",
+                        this.label, flowType, line);
             }
         });
         /* stderr */
@@ -248,10 +248,9 @@ function onDestinationSSHConnReady(flowType) {
                     /* Plot an extra iperf point */
                     state.plotter[flowType].stdin.write(
                             time + " " + this.id + " " + bw + "\n");
-                } else {
-                    console.log("%s %s Destination STDOUT: %s",
-                            this.label, flowType, line);
                 }
+                console.log("%s %s Destination STDOUT: %s",
+                        this.label, flowType, line);
             } else if (flowType == "ping") {
                 /* PIT measurements taken by tcpdump. */
                 var words = line.trim().split(/[, ]+/);
-- 
2.7.4
````

echizenr, 2018-09-19:
I apply the patch.

First test I was "lucky" because the full error happen (board 2 to 3 flow started and stop just after few second and kill the LBT process) : 

> {"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},"destination":{"user":
> "root","hostname":"172.15.0.3","port":22},"port":5201,"transport":"tcp","bandwidth":-1,"enabled":true,"
> label":"Iperf (Board 1 -> Board 3)","id":"fefcdf3b-d73a-4129-a0ea-046d8bb90334"},{"source":{"user":"roo
> t","hostname":"192.168.15.2","port":22},"destination":{"user":"root","hostname":"172.15.0.3","port":22}
> ,"port":5202,"transport":"tcp","bandwidth":-1,"enabled":false,"label":"Iperf (Board 2 -> Board 3)","id"
> :"2b2f4975-efc4-4371-9338-e4814ee0afea"}],"ping":[]}}
> Successfully written flows to file.
> PUT /running
> traffic start/stop: old state false, new state true
> sse :: established new connection to ::ffff:192.168.15.100
> Iperf (Board 1 -> Board 3) iperf Destination :: conn ready
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
> ---------
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
> ---------
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: Server listening on 5201
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
> ---------
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
> ---------
> Iperf (Board 1 -> Board 3) iperf Client :: conn ready
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: Connecting to host 172.15.0.3, port 5201
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 50644
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.1, port 50644
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [  4] local 172.15.0.1 port 50646 connected to 172.1
> 5.0.3 port 5201
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5201 connected to 172.
> 15.0.1 port 50646
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5201 connected to 172.
> 15.0.1 port 50646
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [ ID] Interval           Transfer     Bandwidth
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   0.00-0.50   sec  47.8 MBytes   801 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   0.50-1.00   sec  41.8 MBytes   701 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [ ID] Interval           Transfer     Bandwidth     
>   Retr  Cwnd
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [  4]   0.00-1.00   sec  91.3 MBytes   763 Mbits/sec
>     0    399 KBytes       
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   1.00-1.50   sec  40.1 MBytes   673 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   1.50-2.00   sec  40.2 MBytes   675 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [  4]   1.00-2.00   sec  80.0 MBytes   674 Mbits/sec
>     0    399 KBytes       
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   2.00-2.50   sec  42.8 MBytes   717 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   2.50-3.00   sec  50.5 MBytes   848 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [  4]   2.00-3.01   sec  94.1 MBytes   782 Mbits/sec
>     0    421 KBytes       
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]   3.00-3.50   sec  53.4 MBytes   895 Mbits/s
> ec                  
> PUT /running
> traffic start/stop: old state true, new state false
> GET /running
> GET /flows
> feedgnuplot process exited with code 0
> Command "iperf3 -t 86400 -p 5201  -c 172.15.0.3" on root@192.168.15.1:22 exited with code undefined and
>  signal undefined
> Command "iperf3 -1 -f m -i 0.5 -s -p 5201" on root@172.15.0.3:22 exited with code undefined and signal 
> undefined
> GET /flows
> GET /running
> PUT /flows
[... 676 lines cut ...]
> Iperf (Board 2 -> Board 3) iperf Destination STDOUT: [  5]  11.00-11.50  sec  13.1 MBytes   220 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Destination STDOUT: [  5]  11.50-12.00  sec  42.8 MBytes   718 Mbits/s
> ec                  
> Iperf (Board 1 -> Board 3) iperf Source :: STDOUT: [  4]  11.01-12.00  sec  85.0 MBytes   719 Mbits/sec
>     0    370 KBytes       
> Iperf (Board 2 -> Board 3) iperf Destination STDOUT: [  5]  11.50-12.00  sec  13.4 MBytes   224 Mbits/s
> ec                  
> PUT /running
> traffic start/stop: old state true, new state false
> feedgnuplot process exited with code 0
> Command "iperf3 -1 -f m -i 0.5 -s -p 5202" on root@172.15.0.3:22 exited with code undefined and signal 
> undefined
> Command "iperf3 -1 -f m -i 0.5 -s -p 5201" on root@172.15.0.3:22 exited with code undefined and signal 
> undefined
> Command "iperf3 -t 86400 -p 5201  -c 172.15.0.3" on root@192.168.15.1:22 exited with code undefined and
>  signal undefined
> Command "iperf3 -t 86400 -p 5202  -c 172.15.0.3" on root@192.168.15.2:22 exited with code undefined and
>  signal undefined
> GET /flows
> GET /running
> GET /flows
> GET /running

It would seem that Iperf 2 flow (Board 2 to 3) worked but LBT is not able to display graph.

vladimiroltean, 2018-09-19:
I don't see anything out of the ordinary in the logs you attached.

Please revert all patches on server.js applied so far and use:

* This as a fix for the stack trace:

```
From 5e53b9334c65b27c4d659260af428fccb416b35c Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Wed, 19 Sep 2018 18:43:34 +0300
Subject: [PATCH 1/3] nodejs-lbt: initialize the plotter early in startFlows()

* It can happen that the stopTraffic() function is called before
  the plotter even has time to initialize, since that was previously
  happening *after* the SSH connections were initiated.
* Start the plotter early, so that killing it is always a valid
  operation in stopTraffic().
* Avoids:

  /usr/lib/node_modules/lbt/server.js:450
  state.plotter[flowType].stdin.end();
  ^

  TypeError: Cannot read property 'end' of undefined
  at forEach (/usr/lib/node_modules/lbt/server.js:450:42)
  at Array.forEach (native)
  at stopTraffic (/usr/lib/node_modules/lbt/server.js:443:23)
  at Channel.stream.on (/usr/lib/node_modules/lbt/server.js:230:17)
  at emitOne (events.js:101:20)
  at Channel.emit (events.js:188:7)
  at Channel. (/usr/lib/node_modules/lbt/node_modules/ssh2/lib/Channel.js:135:18)
  at Channel.g (events.js:292:16)
  at emitNone (events.js:91:20)
  at Channel.emit (events.js:185:7)

---
 package/nodejs-lbt/src/server.js | 40 ++++++++++++++++++++--------------------
 1 file changed, 20 insertions(+), 20 deletions(-)

diff --git a/package/nodejs-lbt/src/server.js b/package/nodejs-lbt/src/server.js
index ddcbed5..5448bbe 100755
--- a/package/nodejs-lbt/src/server.js
+++ b/package/nodejs-lbt/src/server.js
@@ -346,6 +346,26 @@ function startFlows(flows, flowType) {
         feedgnuplotParams.push("--binwidth",  config[flowType].binwidth);
     }
     /* "--timefmt", "%H:%M:%S", "--set", 'format x "%H:%M:%S"', */
+    var plotter = spawn("feedgnuplot", feedgnuplotParams);
+    plotter.stdout.setEncoding("utf8");
+    plotter.stderr.setEncoding("utf8");
+    /* feedgnuplot stdout is delimited by </svg> ending tag.
+     * No need to emit line events. */
+    plotter.stdout.on("data", (data) => onGnuplotData.call(plotter, flowType, data));
+    /* Parse stderr of feedgnuplot by lines */
+    readline.createInterface({ input: plotter.stderr })
+    .on("line", (line) => {
+        console.log("feedgnuplot stderr: %s", line);
+        /* Some warning messages are not harmful */
+        if (!line.includes("adjusting to")) {
+            stopTraffic();
+        }
+    });
+    plotter.on("exit", (code) => {
+        console.log("feedgnuplot process exited with code %s", code);
+    });
+    plotter.svg = "";
+    state.plotter[flowType] = plotter;
 
     state.startTime = Date.now();
     flows.forEach((f) => {
[... 83 lines cut ...]
         state.clients.forEach((stream) => {
+            fs.writeFile("plot" + this.counter + ".svg", this.svg, () => {
+                console.log("plot ", this.counter, " written to file.");
+            });
             stream.send(flowType, JSON.stringify({ svg: this.svg }));
         });
         /* Re-initialize the svg with the remainder */
         this.svg = halves[1];
+        this.counter = this.counter + 1;
     } else {
         this.svg += data;
     }
@@ -352,6 +356,7 @@ function startFlows(flows, flowType) {
     plotter.stderr.setEncoding("utf8");
     /* feedgnuplot stdout is delimited by </svg> ending tag.
      * No need to emit line events. */
+    plotter.counter = 0;
     plotter.stdout.on("data", (data) => onGnuplotData.call(plotter, flowType, data));
     /* Parse stderr of feedgnuplot by lines */
     readline.createInterface({ input: plotter.stderr })
-- 
2.7.4
```

Also, I forgot to ask, but what browser version were you using before, and what browser are you using now? I think this application has only been tested on certain versions of Firefox.

echizenr, 2018-09-20:
> Also, I forgot to ask, but what browser version were you using before, and what browser are you using now? I think this application has only been tested on certain versions of Firefox.

I'm currently use Firefox  64 bits v62.0

I will try to apply those patch and come back to you with new logs

vladimiroltean, 2018-09-20:
Ok. Just send me logs from the failing case (Board 2 -> Board 3) this time.

echizenr, 2018-09-20:
here you can find logs for Iperf test (Board 2 -> Board 3).

After being stuck on main page for approximately 10 seconds, LBT display a graph but the plot of Iperf flow was halting and after 2 sequence value ploted, nothing happened. Moreover, when I stopped the tests, the LBT is closed itself.

```
GET /
GET /client.js
GET /flows
GET /running
PUT /flows
{"flows":{"iperf":[{"source":{"user":"root","hostname":"192.168.15.1","port":22},"destination":{"user":
"root","hostname":"172.15.0.3","port":22},"port":5201,"transport":"tcp","bandwidth":-1,"enabled":false,
"label":"Iperf (Board 1 -> Board 3)","id":"5bfa1143-2d41-4a12-b1a1-102ad61f0099"},{"source":{"user":"ro
ot","hostname":"192.168.15.2","port":22},"destination":{"user":"root","hostname":"172.15.0.3","port":22
},"port":5202,"transport":"tcp","bandwidth":-1,"enabled":true,"label":"Iperf (Board 2 -> Board 3)","id"
:"df603e6d-8800-4d3b-bdf1-f853f593b2e7"}],"ping":[]}}
Successfully written flows to file.
PUT /running
traffic start/stop: old state false, new state true
sse :: established new connection to ::ffff:192.168.15.100
Iperf (Board 2 -> Board 3) iperf Destination :: conn ready
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
---------
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
---------
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: Server listening on 5202
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
---------
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: --------------------------------------------------
---------
Iperf (Board 2 -> Board 3) iperf Client :: conn ready
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: Connecting to host 172.15.0.3, port 5202
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 53362
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: Accepted connection from 172.15.0.2, port 53362
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4] local 172.15.0.2 port 53364 connected to 172.1
5.0.3 port 5202
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5202 connected to 172.
15.0.2 port 53364
Iperf (Board 2 -> Board 3) iperf Destination STDOUT: [  5] local 172.15.0.3 port 5202 connected to 172.
15.0.2 port 53364
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [ ID] Interval           Transfer     Bandwidth     
  Retr  Cwnd
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   0.00-1.00   sec   110 MBytes   924 Mbits/sec
    0    365 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   1.00-2.00   sec   113 MBytes   943 Mbits/sec
    0    365 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   2.00-3.00   sec   113 MBytes   946 Mbits/sec
    0    783 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   3.00-4.00   sec   112 MBytes   940 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   4.00-5.00   sec   113 MBytes   945 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   5.00-6.00   sec   112 MBytes   942 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   6.00-7.00   sec   112 MBytes   938 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   7.00-8.00   sec   113 MBytes   945 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   8.00-9.00   sec   112 MBytes   940 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]   9.00-10.00  sec   112 MBytes   940 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  10.00-11.00  sec   112 MBytes   943 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  11.00-12.00  sec   112 MBytes   937 Mbits/sec
    0    822 KBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  12.00-13.00  sec   111 MBytes   932 Mbits/sec
    0   1.01 MBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  13.00-14.00  sec   103 MBytes   867 Mbits/sec
    0   1.01 MBytes       
[... 375 lines cut ...]
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  62.00-63.00  sec   111 MBytes   936 Mbits/sec
    0   1.63 MBytes       
Iperf (Board 2 -> Board 3) iperf Source :: STDOUT: [  4]  63.00-64.00  sec   111 MBytes   933 Mbits/sec
    0   1.63 MBytes       
PUT /running
traffic start/stop: old state true, new state false
/usr/lib/node_modules/lbt/server.js:457
         state.plotter[flowType].stdin.end();
                                      ^

TypeError: Cannot read property 'end' of undefined
    at forEach (/usr/lib/node_modules/lbt/server.js:457:32)
    at Array.forEach (native)
    at stopTraffic (/usr/lib/node_modules/lbt/server.js:454:23)
    at onStartStopTraffic (/usr/lib/node_modules/lbt/server.js:494:9)
    at IncomingMessage.request.on.e (/usr/lib/node_modules/lbt/server.js:91:17)
    at emitNone (events.js:86:13)
    at IncomingMessage.emit (events.js:185:7)
    at endReadableNT (_stream_readable.js:974:12)
    at _combinedTickCallback (internal/process/next_tick.js:80:11)
    at process._tickCallback (internal/process/next_tick.js:104:9)

[1]+  Done(1)                    /usr/lib/node_modules/lbt/server.js

```

vladimiroltean, 2018-09-20:
I see that the "iperf Destination STDOUT" emitted some data rather sporadically, a small burst around 15 to 18 seconds in, and a second burst around the 32 seconds timestamp.
I'm afraid there's not much the LBT program can do to help with that. Basically there's a network issue with the SSH connection to the destination, which prevents the iperf stdout from arriving (and being plotted) in time and not bursty.
I still stand by my position that the login IP and the traffic IP should be different. In fact, below you can find attached a patch where these are separated. You have to delete your existing flows.json and re-create the setup. Please configure the "Source IP" and "Destination IP" in the 172.15.0.x network, and "Source Login" and "Destination Login" in the 192.168.15.x network.

```
From 20d2999765c0a5588fb2238115b575c5cb50898a Mon Sep 17 00:00:00 2001
From: Vladimir Oltean <vladimir.oltean@nxp.com>
Date: Thu, 20 Sep 2018 15:04:07 +0300
Subject: [PATCH] nodejs-lbt: separate the networks for login and data

* This allows iperf and ping traffic to be ran on different routes
  than the network route used to establish an SSH session between the
  LBT web server and those machines.

---
 package/nodejs-lbt/src/client.js  | 32 +++++++++++++++++++++++++-------
 package/nodejs-lbt/src/flows.json | 10 ++++++++++
 package/nodejs-lbt/src/index.html | 10 +++++++---
 package/nodejs-lbt/src/server.js  | 20 ++++++++++++++------
 4 files changed, 56 insertions(+), 16 deletions(-)

diff --git a/package/nodejs-lbt/src/client.js b/package/nodejs-lbt/src/client.js
index 6f554ad..fe21c3c 100644
--- a/package/nodejs-lbt/src/client.js
+++ b/package/nodejs-lbt/src/client.js
@@ -14,7 +14,11 @@ var serverState = {
 /* Type of e is InputEvent.
  * Type of e.target is HTMLTableCellElement. */
 function changeFlow(classes, text) {
-    if (classes.contains("source")) {
+    if (classes.contains("sourceLogin")) {
+        this.sourceLoginText = text;
+    } else if (classes.contains("destinationLogin")) {
+        this.destinationLoginText = text;
+    } else if (classes.contains("source")) {
         this.sourceText = text;
     } else if (classes.contains("destination")) {
         this.destinationText = text;
@@ -45,8 +49,10 @@ function addFlow(flowType) {
     switch (flowType) {
     case "iperf":
         this.push({
-            sourceText: "user@hostname:port",
-            destinationText: "user@hostname:port",
+            sourceText: "hostname",
+            sourceLoginText: "user@hostname:port",
+            destinationText: "hostname",
+            destinationLoginText: "user@hostname:port",
             port: "n/a",
             transport: "tcp",
             bandwidth: "n/a",
@@ -56,8 +62,10 @@ function addFlow(flowType) {
         break;
     case "ping":
         this.push({
-            sourceText: "user@hostname:port",
-            destinationText: "user@hostname:port",
+            sourceText: "hostname",
+            sourceLoginText: "user@hostname:port",
+            destinationText: "hostname",
+            destinationLoginText: "user@hostname:port",
             intervalType: "adaptive",
             intervalMS: "n/a",
             packetSize: "64",
@@ -93,6 +101,8 @@ function populateRow(flowType, flow) {
     var label           = '<td ' + index + inputEditable + ' class="editable label">' + flow.label + '</td>';
     var sourceText      = '<td ' + index + inputEditable + ' class="editable source">' + flow.sourceText + '</td>';
     var destinationText = '<td ' + index + inputEditable + ' class="editable destination">' + flow.destinationText + '</td>';
+    var sourceLoginText = '<td ' + index + inputEditable + ' class="editable sourceLogin">' + flow.sourceLoginText + '</td>';
+    var destinationLoginText = '<td ' + index + inputEditable + ' class="editable destinationLogin">' + flow.destinationLoginText + '</td>';
[... 159 lines cut ...]
  *                 intervalMS: integer,
  *                 enabled: boolean,
@@ -603,7 +607,9 @@ function curateStateForSend(state) {
     state.flows.iperf.forEach((f) => {
         newFlows.iperf.push({
             source: f.source,
+            sourceLogin: f.sourceLogin,
             destination: f.destination,
+            destinationLogin: f.destinationLogin,
             port: f.port,
             transport: f.transport,
             bandwidth: f.bandwidth,
@@ -614,7 +620,9 @@ function curateStateForSend(state) {
     state.flows.ping.forEach((f) => {
         newFlows.ping.push({
             source: f.source,
+            sourceLogin: f.sourceLogin,
             destination: f.destination,
+            destinationLogin: f.destinationLogin,
             intervalType: f.intervalType,
             intervalMS: f.intervalMS,
             packetSize: f.packetSize,
-- 
2.7.4
```

echizenr, 2018-09-20:
Thank you, I will try your solution but I still don't understand why it worked before and no longer. Because I doesn't change anything except xml files and update sdcard.img for new kernel.

vladimiroltean, 2018-09-20:
You said this also occurs with the prioritizing xml. But if you take a moment to understand what the prioritizing config does, you'll see that in the case of a single flow it actually behaves identically to the standard configuration. You only notice the prioritization when you have multiple flows, one with a higher priority than the other (that is to say, priority is relative).
So I'd ask you to first confirm that the bursty iperf data between board 2 -> board 3 happens with a clean config (`sja1105-tool config default -f ls`), and then focus on that scenario only until it's solved.

echizenr, 2018-09-20:
Here you can find the manual Iperf test I made, It seems to work : 

```
**Server**

iperf -s -p 5201
------------------------------------------------------------
Server listening on TCP port 5201
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 172.15.0.3 port 5201 connected with 172.15.0.2 port 57092
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.0 sec  1.09 GBytes   939 Mbits/sec

**Client**

iperf -c 172.15.0.3 -p 5201
------------------------------------------------------------
Client connecting to 172.15.0.3, TCP port 5201
TCP window size: 43.8 KByte (default)
------------------------------------------------------------
[  3] local 172.15.0.2 port 57092 connected with 172.15.0.3 port 5201
[ ID] Interval       Transfer     Bandwidth
[  3]  0.0-10.0 sec  1.09 GBytes   940 Mbits/sec
```

echizenr, 2018-11-22:
Thank you for coming back to me. I checked your previous answer and modified few files in order to seperate IP source & Destination and IP login & Destination.

Fun fact: after a complete rebuild of sdcard.img after a few weeks, this problem was solved without any changes compared to those made when the problem occurred, but I still made the changes you made. have advised, in order to have a robust version against this problem.

I'm currently working on script modification in order to have more than 2 Iperf flow allowed during TSN demo tests. If your are interested, I could create a topic where I could display the results.

vladimiroltean, 2018-11-22:
Thank you, feel free to contribute back fixes or improvements to the demo applications.

echizenr, 2018-11-23:
Ok, I will continue my tests and I will open an issue which purpose will be to discuss around those results.

On another level, regarding a previous question related to TSN functionalities evolutions, you told me that Frame preemption on the Ethernet controller ports and Frame preemption and seamless redundancy on the integrated L2 switch will be available on LS1028 chips. Could you provide me a COTS roadmap of NXP's involvement & evolutions on TSN if you have one ?

vladimiroltean, 2018-12-03:
> Could you provide me a COTS roadmap of NXP's involvement & evolutions on TSN if you have one ?

Sorry for acknowledging your request so late. I still don't have an LS1028 roadmap that I can share unfortunately, but I'll come back to you when I do.

echizenr, 2018-12-04:
No problem, I'm sure you have lot of work to do and I'm not the only one who request your help !

Ok, Thank you. 
To be more specific, I need a roadmap which target NXP TSN chips, IP, core and all you component that are TSN compliant. 
In parallel, having NXP referents, we will organize a meeting to discuss this subject.

I thank you by advance for information you will provide me.

closed 2019-03-10 by vladimiroltean