nxp-archive/openil_sja1105-tool #25: Trying to initialize l2-address-lookup-parameter-table crashes

sylane, 2018-10-02:
Trying to initialize the `l2-lookup-parameters` table make the command line tool crash:

```
# sja1105-tool config save config.xml`
```

Update config.xml:
```
               <l2-address-lookup-parameters-table>
                        <entry>
                                <index>0</index>
                                <maxage>0</maxage>
                                <dyn_tbsz>0</dyn_tbsz>
                                <poly>0x97</poly>
                                <shared_learn>0</shared_learn>
                                <no_enf_hostprt>0</no_enf_hostprt>
                                <no_mgmt_learn>1</no_mgmt_learn>
                        </entry>
                </l2-address-lookup-parameters-table>
```
```
# sja1105-tool config load config.xml
*** Error in `sja1105-tool': double free or corruption (!prev): 0x000600d0 ***
Aborted
```

Trying to modify the parameter using the command-line tools doesn't work either:

```
# sja1105-tool config modify                                               
sja1105-tool config modify <table-name>[<entry-index>] <field-name> <field-value>
Please run "sja1105-tool config modify help" to see more details
# sja1105-tool config modify l2-address-lookup-parameters-table0 dyn_tbsz 0
Token "l2-address-lookup-parameters-table0" matches no known command. Try:
   * schedule-table
   * schedule-entry-points-table
   * vl-lookup-table
   * vl-policing-table
   * vl-forwarding-table
   * l2-address-lookup-table
   * l2-policing-table
   * vlan-lookup-table
   * l2-forwarding-table
   * mac-configuration-table
   * schedule-parameters-table
   * schedule-entry-points-parameters-table
   * vl-forwarding-parameters-table
   * l2-address-lookup-parameters-table
   * l2-forwarding-parameters-table
   * clock-synchronization-parameters-table
   * avb-parameters-table
   * general-parameters-table
   * retagging-table
   * xmii-mode-parameters-table
   * sgmii-table
# sja1105-tool config modify l2-address-lookup-parameters-table dyn_tbsz 0
Index out of bounds!
Please adjust the entry count of the table:
* config modify <table> entry-count <value>)
modify failed!
# sja1105-tool config modify l2-address-lookup-parameters-table entry-count 1
*** Error in `sja1105-tool': double free or corruption (out): 0x0003a5c0 ***
Aborted
```

Adding entries to other tables seems to work:

```
# sja1105-tool config show avb-parameters-table
Audio/Video Bridging Parameters Table is empty
# sja1105-tool config modify avb-parameters-table entry-count 1
# sja1105-tool config show avb-parameters-table
Audio/Video Bridging Parameters Table: 1 entries
|| Entry 0:                            || 
|| DESTMETA 00:00:00:00:00:00          || 
|| SRCMETA  00:00:00:00:00:00          || 
||                                     || 
```

sylane, 2018-10-02:
The PR #26 is fixing the crash.
But it is still not possible to set the values of `l2-address-lookup-parameters-table`.
Adding it to the XML config and loading it works but then the table is still empty afterward.
Trying to modify the table do not work either:

```
# sja1105-tool config load -f default.xml
# sja1105-tool config show l2-address-lookup-parameters-table
L2 Address Lookup Parameters Table is empty
# sja1105-tool config load -f disable-learning-config.xml
# sja1105-tool config show l2-address-lookup-parameters-table
L2 Address Lookup Parameters Table is empty
# sja1105-tool config modify l2-address-lookup-parameters-table dyn_tbsz 0
Index out of bounds!
Please adjust the entry count of the table:
* config modify <table> entry-count <value>)
modify failed!
# sja1105-tool config show l2-address-lookup-parameters-table
L2 Address Lookup Parameters Table is empty
# sja1105-tool config modify l2-address-lookup-parameters-table entry-count 1
# sja1105-tool config show l2-address-lookup-parameters-table
There can be no more than 1 L2 Lookup Parameters entries (2 present)
error while interpreting config
# sja1105-tool config modify l2-address-lookup-parameters-table dyn_tbsz 0
There can be no more than 1 L2 Lookup Parameters entries (2 present)
error while interpreting config
```

vladimiroltean, 2018-10-03:
Thanks for providing disable-learning-config.xml. Save/load/modify operations work with the change I added to your pull request.

sylane, 2018-10-04:
Thank you, the configuration loads fine now... Sadly it still doens't disable the switch auto-learning :(

vladimiroltean, 2018-10-04:
Did you turn off the per-MAC DYN_LEARN field?

sylane, 2018-10-04:
I do:

sylane, 2018-10-04:
You are 100% right ! I used wireshark and the traffic is indeed forwarded everywhere !
Now I only need to have the frames for unknown address to be sent only to the host port !
Thank you.

sylane, 2018-10-04:
The `FL_DOMAIN` works perfectly to redirect the frames for unknown address only to the host, thank you for the pointer.

closed 2018-10-04 by sylane