nxp-archive/openil_sja1105-tool #27: The l2-address-lookup-table cannot be modified on E/T

sylane, 2018-10-04:
From the documentation `UM10944 Software user manual for SJA1105TEL`, on the E/T, the `l2-address-lookup-table` only have the fields `VLANID`, `MACADDR`, DESTPORTS`, ENFPORT` and  `INDEX`.

`sja1105-tool` requires the fields `tsreg`, `mirrvlan`, `takets`, `mirr`, `retag`, `mask_iotag`, `mask_vlanid`, `mask_macaddr` and `iotag`.

vladimiroltean, 2018-10-05:
Not really a bug. Those additional fields you see in `sja1105-tool config modify l2-address-lookup-table` are for the P/Q/R/S series. They are kept in a common structure and unpacked into the binary staging area based on device ID: https://github.com/openil/sja1105-tool/blob/master/src/tool/config-modify.c#L165-L224
You should be able to read/write the fields that E/T support with no problem.

sylane, 2018-10-05:
You say I just need to put dummy data in the XML files ? because if I don't specify these fields the tool refuse to load the file.

vladimiroltean, 2018-10-05:
Oh, ok, I didn't think about that.
You could either put dummy data, or you should be able to add some conditionals based on `config->device_id` in `src/tool/xml/write/l2-lookup.c` and `src/tool/xml/read/l2-lookup.c` so it doesn't attempt to access the extra data at all.

closed 2019-05-06 by vladimiroltean