nxp-archive/openil_sja1105-tool #28: Hi. How do I configure the eth2, eth3, eth4, eth5 ports using sja1105-tool? I understood that I can't configure these ports using Linux as they don't have have a Linux kernel. I am working on LS1021ATSN board. Kindly help. Thanks

rubidebnath, 2018-10-11:


rubidebnath, 2018-10-11:
Hi. I am working on LS1021ATSN board. I understood that I can't configure eth2, eth3, eth4, eth5 ports using Linux CLI as they don't have a Linux kernel. So, how to configure these ports using sja1105-tool? Kindly help. Thanks

vladimiroltean, 2018-10-11:
I assume you are talking about the ports labeled with capital ETH2 - ETH5 on the chassis, because eth3, eth4, eth5 do not exist on this board.
What do you want to configure about them?

vladimiroltean, 2018-10-11:
Does your host have a 100Mbps or a 1Gbps Ethernet port?
If 100, you need to modify `/etc/init.d/S46sja1105-link-speed-fixup`, set `ETH5_SPEED_MBPS=100` and then either re-run the script or reboot the board.

rubidebnath, 2018-10-11:
I modified the script but now also I am not able to ping the host. Also whenever I do ifconfig there is only eth0, eth1, eth2. Why the rest other ports are not visible? 

[root@OpenIL:~]# ifconfig -a
eth0      Link encap:Ethernet  HWaddr 00:04:9F:EF:06:06  
          inet addr:192.168.15.1  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:feef:606/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:252 errors:0 dropped:0 overruns:0 frame:0
          TX packets:64 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:46555 (45.4 KiB)  TX bytes:10284 (10.0 KiB)
          Base address:0xe000 

eth1      Link encap:Ethernet  HWaddr 00:04:9F:EF:07:07  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
          Base address:0xc000 

eth1:avahi Link encap:Ethernet  HWaddr 00:04:9F:EF:07:07  
          inet addr:169.254.5.79  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          Base address:0xc000 

eth2      Link encap:Ethernet  HWaddr 00:04:9F:EF:08:08  
          inet addr:172.15.0.1  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:feef:808/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:141 errors:0 dropped:0 overruns:0 frame:0
          TX packets:76 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:26099 (25.4 KiB)  TX bytes:12372 (12.0 KiB)
          Base address:0xa000 

eth2.100  Link encap:Ethernet  HWaddr 00:04:9F:EF:08:08  
          inet addr:172.15.100.1  Bcast:0.0.0.0  Mask:255.255.255.0
          inet6 addr: fe80::204:9fff:feef:808/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:33 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:5432 (5.3 KiB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:49 errors:0 dropped:0 overruns:0 frame:0
          TX packets:49 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:4312 (4.2 KiB)  TX bytes:4312 (4.2 KiB)

sit0      Link encap:IPv6-in-IPv4  
          NOARP  MTU:1480  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

How do I make the ports ETH3-5 up?

vladimiroltean, 2018-10-11:
What IP address are you trying to ping, and what IP address is assigned to your host?

> How do I make the ports ETH3-5 up?

The switch ports are brought up by the `/etc/init.d/S45sja1105` init script.

rubidebnath, 2018-10-11:
Could you pls elaborate on what exactly I have to do to bring the switch ports up?  I want to assign some IP address to the switch ports as well. So, it is important for me to know how exactly I can do that.

vladimiroltean, 2018-10-11:
The switch is a switch, not an L3 entity. You can't assign IP addresses to the switch ports, since fundamentally L3 traffic termination on the switch wouldn't make sense.
The only thing you can do is ping network devices connected to the switch. On the Layerscape side, that device is the `eth2` interface which is directly connected to port 4 of the switch on the PCB. Pinging 172.15.0.1 should work if you configure your host to the same subnet.

vladimiroltean, 2018-10-11:
> Could you pls elaborate on what exactly I have to do to bring the switch ports up?

Nothing, since it's done at boot time. You only need to adjust the link speed per port if you intend to connect 100Mbps devices.

rubidebnath, 2018-11-08:
Thanks a lot for your time and answers. The issue is solved now. Many Thanks.

closed 2018-11-08 by rubidebnath