nxp-archive/openil_sja1105-tool #30: Loading the configuration to the JSA1105T doesn't work the first time.

sylane, 2018-10-23:
I am investigating the configuration for the JSA1105T on the LS1021ATSN-PA kit.
I have a vlan configured and I am changing the `TAG_PORT` field for it in the `vlan-lookup-table`.
I am verifying that the VLAN ID tags are removed, when I set the falue to `0x1F` the tag is not removed, as expected, but when I set it to `0x0`, the first time I load the configuration by doing:

```
> sja1105-tool reset cold

vladimiroltean, 2018-10-23:
In the failing situation (so before running the command a second time), what is printed for `TAG_PORT` if you run `sja1105-tool config show`? 0 or 1f?

sylane, 2018-10-23:
```

vladimiroltean, 2018-10-23:
So you're saying that VLAN Lookup Table starts having 2 entries instead of 1 out of nowhere?
Could you try to print the value of `staging_area.static_config.vlan_lookup_count` throughout this area? https://github.com/openil/sja1105-tool/blob/master/src/tool/parse-config.c#L96-L115
I would need to know at what stage is the invalid value introduced.

sylane, 2018-10-23:
```
[root@etsn-board:local]# sja1105-tool config load -f /tmp/spbv-default.xml 
config_parse_args LINE  96 - vlan_lookup_count: 1
config_parse_args LINE  98 - vlan_lookup_count: 1
config_parse_args LINE 104 - vlan_lookup_count: 1
config_parse_args LINE 110 - vlan_lookup_count: 1
[root@etsn-board:local]# sja1105-tool config show vlan-lookup-table
VLAN Lookup Table: 1 entries
|| Entry 0:             || 
|| VING_MIRR  0x0       || 
|| VEGR_MIRR  0x0       || 
|| VMEMB_PORT 0x1F      || 
|| VLAN_BC    0x1F      || 
|| TAG_PORT   0x1F      || 
|| VLANID     0x800     || 
||                      || 
[root@etsn-board:local]# vi /tmp/spbv-default.xml 
[root@etsn-board:local]# cat /tmp/spbv-default.xml | grep tag_port
				<tag_port>0x0</tag_port>
[root@etsn-board:local]# sja1105-tool config load -f /tmp/spbv-default.xml 
config_parse_args LINE  96 - vlan_lookup_count: 1
config_parse_args LINE  98 - vlan_lookup_count: 1
config_parse_args LINE 104 - vlan_lookup_count: 1
config_parse_args LINE 110 - vlan_lookup_count: 1
[root@etsn-board:local]# sja1105-tool config show vlan-lookup-table
VLAN Lookup Table: 2 entries
|| Entry 0:             || Entry 1:             || 
|| VING_MIRR  0x0       || VING_MIRR  0x0       || 
|| VEGR_MIRR  0x0       || VEGR_MIRR  0x0       || 
|| VMEMB_PORT 0x1F      || VMEMB_PORT 0x11      || 
|| VLAN_BC    0x1F      || VLAN_BC    0x1F      || 
|| TAG_PORT   0x0       || TAG_PORT   0x1E      || 
|| VLANID     0x800     || VLANID     0x800     || 
||                      ||                      || 
```
The line number are where I added a `printf`, it therefor increment the following line numbers...

sylane, 2018-10-31:
I got this with vlan-lookup-table too and this is a major issue for us...
When loading a configuration with a single vlan entry in the table it create 2 entries the first time `sja1105-tool config load -f file.xml` is called.

sylane, 2018-10-31:
Even when resetting the table size before loading this generate two entries:

sylane, 2018-10-31:
And we cannot even use the tool to modify the config field-by-field !
```

vladimiroltean, 2018-10-31:
On the last point here, you got the syntax wrong, try the following: `sja1105-tool config modify vlan-lookup-table[1] vlanid 123`.
On the bug: when you run `sja1105-tool config load ./config2.xml` and it works (the VLAN lookup table has 1 entry), is the staging area any different than when you run the same command and it doesn't (table has 2 entries)?

sylane, 2018-11-01:
For this particular build I used linaro tool-chain exactly as described in the readme.
The behavior is 100% reproducible. Every time I am loading a different config with a single entry (not tested with more than 1 entry yet) it duplicate the entries and the switch is no behaving as expected.
Even worth, on our prototype we programmatically call the tool when something need to change, and even calling it two times with the same config is unreliable...

vladimiroltean, 2018-11-01:
What about my other questions?

sylane, 2018-11-01:
ok, I rebuilt the tool from scratch and I can't seem to reproduce the issue... weird...

vladimiroltean, 2018-11-01:
Any chance it was a mismatch betweenb lib and tool versions?

sylane, 2018-11-01:
It is possible, but I cannot be sure...

vladimiroltean, 2018-11-01:
If you can't be sure, can I consider this issue closed?

sylane, 2018-11-01:
Yes, if I see this issue again I'll open a new issue.

closed 2018-11-01 by sylane