nxp-archive/openil_sja1105-tool #31: Static routes doesn't seem to work on the SJA1105T

sylane, 2018-10-23:
I am trying to test adding static routing using the `l2-address-lookup-table` on the LS1021ATSN-PA kit.

Because I didn't figure out the crc algorithm used to get an index grom a mac address yet I just added 255 entries with index from 0 to 1020 with an increment of 4 (0, 4, 8, 12...) so I feel on entry for each possible hash.

The entries all have the same MAC address (`01:02:03:L04:05:06`) and a `0x4` as `destports` (ETH3).

When sending a frame with destination address `01:02:03:L04:05:06`, the frame is forwarded to all ports, not only ETH3. That leads me to assume that this is not working.

I see in the documentation about `l2-address-lookup-table`: `loading of this block must not start before the L2BUSYS flag in the status area has been cleared`. I coudn't find in the code if the tool was actually checking for this flag.

Any idea why l2 address lookup wouldn't work ? Any idea ion how to debug this issue ?

Thank you.

vladimiroltean, 2018-10-23:
Do you have an XML configuration file with the setup done so far?

sylane, 2018-10-23:
This is the test configuration I was talking about: https://gist.github.com/sylane/eb250110770e490cacb52e565e20a1fa

sylane, 2018-10-23:
I got management traffic to be redirected to the host port just fine, but the management traffic going **out** of the host port is dropped with the error `N_POLERR`. So I assumed I needed to add an entry in the `l2-address-lookup-table` for the management traffic mac address. But before spending time figuring out the CRC algorithm used for the hash table I wanted to test it. I couldn't get it to work...

sylane, 2018-10-23:
For what I see it may actually work. I was monitoring the host port and it was receiving my frame, but the other ports (beside the configured one) actually don't receive the frame. The question is why does the host port receive the frame ? I don't see anywhere in my configuration that could explain that.

sylane, 2018-10-23:
And this is definitely caused by the the fact that the destination match the management traffic filter, if I disable the management traffic the frame is routed correctly to the other ports. But then the incoming frames are not taken as management traffic anymore...

vladimiroltean, 2018-10-23:
This portion is not very clear to me either. I will do some experimentation with your config tomorrow and come back when I figure out what's going on.

sylane, 2018-10-23:
This is the last configuration with only the relevant entry for `l2-address-lookup-table`: https://gist.github.com/sylane/45ec4f9d7d3d3275bc0a61c00b7c2108

sylane, 2018-10-25:
Would be nice to have the tool actually generate the correct index from the MAC and the VID. The documentation is far from clear about how to calculate it.

vladimiroltean, 2018-10-25:
> But got the l2-address-lookup-table the index field is a real field that is required for the entry to make sense. Is it a mistake in the documentation or there is an issue with this table ?

Good catch, I missed this completely but luckily the tool does the right thing: for most tables it just writes a dummy value ([the entry's index](https://github.com/openil/sja1105-tool/blob/master/src/tool/xml/write/l2-policing.c#L43)) but for the L2 Address Lookup table it actually [writes the value of the field](https://github.com/openil/sja1105-tool/blob/master/src/tool/xml/write/l2-lookup.c#L56). The value is also [correctly read back](https://github.com/openil/sja1105-tool/blob/master/src/tool/xml/read/l2-lookup.c#L49). So luckily it is just a documentation issue.

I do get your point about the tool being responsible for calculating the index. However I don't exactly know how to do that, maybe you can help.

What I do know is that this formula should hold true:
```
set_index = crc8(vlan_id, dest_mac_addr)
```
so set_index is a value 0-> 255 which points to a 4-way associative entry. You should be able to place the FDB entry at an index anywhere between 4*set_index and 4*set_index + 3.
The vlan_id and dest_mac_addr should be concatenated into an 8-byte word before being fed into the crc8. If SHARED_LEARN is set, I think you still need to do that, but vlan_id has to be set as zeroes.
The crc8 uses the POLY value as a polynomial (in Koopman notation).
This is all I know, I don't have an implementation.

vladimiroltean, 2018-10-25:
I lost you with all the issues opened at once.
I am able to configure the switch using your initial xml to send frames only to Port 2 if the destination MAC is 01:02:03:04:05:06. Are you still not able to do that?

sylane, 2018-10-25:
About the index it is more complicated than that:
  * It depends on the `POLY` fields of the general parameter table.
  * The index is something like `4 * CRC(POLY, MAC, VID) + N` with N in `[0, 1, 2, 3]`.
  * I don't have any idea how the crc is calculated ! The one I found was by setting up all the possible CRC and bisecting until I got the correct index... This is why I would have liked the tool to know how to do it :)

My current issue is that the frames sent to the MAC address `01:80:c2:00:00:14` received on port 0 to 3 are correctly forwarded the the host port 4, **BUT** the frames sent to the MAC address `01:80:c2:00:00:14` received on the host port 4 are dropped.

sylane, 2018-10-25:
I was able to have management traffic to get out of the host port, but this requires executing these commands **for each frames**:

```
    sja1105-tool reg 0x20 0xC7C00000 # ETH5:1, ENFPORT:1, INDEX: 124, PADDING
    sja1105-tool reg 0x21 0x20000147 # MAC1/2, HOST:0, ETH4:1, ETH3:1, ETH2:1
    sja1105-tool reg 0x22 0x0000180C # VID:0, MAC2/2
    sja1105-tool reg 0x23 0xC4000000 # VALID:1, RDWRSET:1, VALIDENT:0, MGMTROUTE:1
```

I am only prototyping for now, so I wasn't expecting to implement a SPI driver, and this sounds heavy to do for every frames...

What would be helpful is for sja1105-tool to have a command line interface (like OpenSSL), then I could run it only once and submit commands...

sylane, 2018-10-26:
This is very nice ! Way more useful with this patch !

vladimiroltean, 2018-11-01:
Did you test out the pull request? Can I merge it and close this issue?

sylane, 2018-11-01:
Sorry, I didn't get time to test it. We are in a crush to finish a prototype by the end of the week and for now we only need one rule we already have an index for.

vladimiroltean, 2018-11-22:
FYI even if you tested the pull request and you can confirm it works, merging it will have to wait until https://github.com/openil/sja1105-tool/pull/36 is ready.

closed 2019-05-06 by vladimiroltean