nxp-archive/openil_sja1105-tool #32: How to dynamically setup a L2 Address Lookup Entry ?

sylane, 2018-10-24:
Trying to resolve the issue with management traffic sent to the host port being dropped, I am trying to setup a "special" management route entry dynamically.

It doesn't seem to work, but I may be doing it wrong (bad bit order?). Is the following the correct way to add an entry ?

 1. Build the binary entry for the L2 Address Lookup table (Table10) for MAC address `01:80:C2:00:00:14`:
```
    PADING:    0000000000000000000
    INDEX:     0000000000
    ENFPORT:   0
    DESTPORTS: 01111
    MACADDR:   000000011000000011000010000000000000000000010100 
    VLANID:    000000000000
```
```
    0b00000000000000000000000000000001111000000011000000011000010000000000000000000010100000000000000
```
```
    0x00000000F0180C2000014000
```

 2. Build the value for the L2 Address Lookup table reconfiguration register (Table 64):

```
    PADDING:   00000000000000000000000000
    MGMTROUTE: 1
    VALIDENT:  0
    LOCKEDS:   0
    ERRORS:    0
    RDWRSET:   1
    VALID:     0
```
```
    0b00000000000000000000000000100010
```
```
    0x00000022
```

Write the registers with the tool:

    sja1105-tool reg 0x20 0x00000000
    sja1105-tool reg 0x21 0xF0180C20
    sja1105-tool reg 0x22 0x00014000
    sja1105-tool reg 0x23 0x00000022

Thank you for your support.

sylane, 2018-10-25:
Is it expected to work that way ?

vladimiroltean, 2018-10-25:
Figuring out bit ordering is no easy stuff unfortunately. For this reason the sja1105-tool actually includes a convenience subsystem called "libgtable": https://github.com/openil/sja1105-tool/tree/master/src/lib/gtable (I recommend viewing the README in an editor with monospaced font, it will make a lot more sense).
The SJA1105 hardware actually uses [QUIRK_LSW32_IS_FIRST](https://github.com/openil/sja1105-tool/blob/master/src/tool/main.c#L184). This means that the 64-bit number 0x0011223344556677 is actually in memory: 44 55 66 77 00 11 22 33. 
You can in fact cross-check with a table whose configuration is supported by the tool, and the output of the `sja1105-tool config hexdump` command.
By the way it is probably easier to write C code than use the raw command.

sylane, 2018-10-25:
I finally got it to work !
I was able to add a route with the command line, I was completely wrong about the bit ordering...

It would be nice to have a command similar to `dump` but to write. Something like:

```
sja1105-tool reg write 0x20 0x87C00000 0x20000140 0x0000180C 0xC8000000
```
That would write all the 32bit values stating at address 0x20 and incrementing the address for each values...

vladimiroltean, 2018-11-22:
FYI even if you tested the pull request and you can confirm it works, merging it will have to wait until https://github.com/openil/sja1105-tool/pull/36 is ready.

vladimiroltean, 2018-11-22:
> It would be nice to have a command similar to dump but to write.

The future of the sja1105-tool is to move to a kernel module backend (see https://github.com/openil/sja1105-tool/pull/36).
There, "raw write" is not something that will be permitted at all, as you can easily change the hardware state outside of the driver's knowledge/control.

vladimiroltean, 2018-11-22:
If you describe your use case clearly, we can think of a way to integrate it nicely with the Linux driver model.

sylane, 2018-11-22:
The use case is:

> Being able to send packets to the MAC address configured as management traffic from the host port.

The only way to do that is to register a temporary management route as described in `5.2.2.1` `Dynamic reconfiguration of the L2 Address Lookup table` field `MGMTROUTE`.

vladimiroltean, 2018-11-22:
It's a bit tricky to inject packets into the switch, because the switch kernel driver needs a handle to the netdevice that acts as host port. The way I understand the issue, it's only a matter of time, there's no other identification of frames performed by the switch. You configure the management route, then the first frame you send on the HOST_PORT of the switch will be treated as management.
An obvious candidate for management route implementation would be in `sja1105-netdev.c` in the `.ndo_start_xmit` callback (currently unsupported). Then you'd be able to send a inject a frame into the switch port of your liking.
But the above interferes really badly with normal operation of the host port. To do that reliably, I assume the sja1105 kmod needs to stop the qdisc queues of the host port netdevice (eth2, in the case of the LS1021A-TSN board), then clone the skb received in `sja1105_xmit` and inject it into the host port's xmit function somehow, then start its qdisc queues back.
Let me know whether you know of a better way of doing this.

vladimiroltean, 2018-11-22:
How do you ensure the management route you configure is for the Ethernet frame you'll be sending?

sylane, 2018-11-22:
Because we send all the management traffic from the same process. This is not 100% safe, other rogue processes could mess things up, but then it means they have access to the service so they could mess up a lot more. Anyway, this is the only option the hardware gives us...

sylane, 2018-11-22:
From what I understand there is 4 management "slots", if we have multiple management handler for different packet we would use different slots for each processes.

vladimiroltean, 2018-11-22:
> different slots for each processes

What do you mean by process? Management traffic identification is done by source MAC (MAC_FLT).

sylane, 2018-11-22:
Mangment traffic is done  by DESTINATION MAC address: 

`Table 22. General Paraeters table` field `MAC_FLT`
```
This field contains a bit mask identifying a bridge level or MAC level management frame
which shall be forwarded only to HOST_PORT. A received L2 frame produces a match if
DEST_MAC and MAC_FLT[i] = MAC_FLTRES[i] holds.
```

sylane, 2018-11-22:
By process, I mean that we have only one process that sends packet with the management traffic MAC address as destination to the different ports. We know which switch port the packet as to go out from, we setup the temporary management route and send the packet. No other process send packets to this destination MAC address.

vladimiroltean, 2018-11-22:
> Mangment traffic is done by **DESTINATION** MAC address

Right, I don't know why I forgot that. Ok, so you can configure up to 4 destinations and you say the risk of collisions for frame injection with other traffic coming from the host port is somewhat lower (and even negligible). If your use-case is sending to a multicast address, I suppose that might be ok for you even without stopping the host port's queues (just take the risk that the host port may send a PDU to that same destination at the same time).
Does this sound less invasive to you then? Egress management traffic would look just like normal xmit through a particular SJA1105 netdevice (there are 5 of those, one per port). Management routes would be configured automatically on the switch for each sent skb (DESTPORTS will be put as the port->index of the netdev). Then the skb will actually be forwarded to the netdevice corresponding to the host port, where real transmission will take place. The SJA1105 driver will find the host port driver through a DTS binding or something.

sylane, 2018-11-22:
Not sure I am following your description. What would be the API/procedure to send a packet from the host port to a specific set of egress port with the destination MAC address being one of the configured management traffic address ?

vladimiroltean, 2018-11-22:
Let me turn that question around and back to you.
I believe what I just said above (management routes to depend on MAC_FLT) to be wrong. That is, MAC_FLT decides what the switch treats as management traffic for ingress (and up to the host port). The management routes are also management traffic, but for switch egress. The MACADDR from the management route entry decides whether egress frames are management or not.
Now my point is: I don't think there's any strict correlation between MAC_FLT and MACADDR from the route. That is to say, there won't be any API for setting the MACADDR. Whatever destination MAC is set in the skb received by sja1105_xmit, that is what the management route gets populated with.
My question to you is: am I right about MAC_FLT being un-correlated to this? Does this sound like something workable for you or am I still missing things?

sylane, 2018-11-22:
I think you make the error in thinking the host port is egress and the other ones are ingress. The host port is just exactly the same as any of the other port, it just append to be the one the cpu is connected to. When sending a packet with a destination address configured as management traffic from the host port, it first pass through the ingress stage on the host port, then it is routed to some other ports and the egress is performed. I think this is why the management route is required, otherwise it could never get out. So to respond to your question, this is completely related to `MAC_FLT`, If I want tos end packets to some mac address that *IS NOT* configured as management traffic it works fine. If the same address matches with `MAC_FLT`/`MAC_FLTRES` the packets are dropped.

vladimiroltean, 2018-11-22:
You're saying that if the mgmt_route.MACADDR & MAC_FLTRES == MAC_FLT, the switch will drop the frame?
You sure about that? Sounds like some sort of misconfiguration on your end, because we do just that for our customized linux-ptp (where we need to process PTP frames - [use MAC_FLT to capture](https://github.com/openil/linuxptp/blob/master/sja1105-ptp-tc.sh#L8), then we modify a correction field, then [we forward back](https://github.com/openil/linuxptp/blob/master/sk.c#L403) to the [same multicast address](https://github.com/openil/linuxptp/blob/master/sja1105-ptp.h#L40)).

vladimiroltean, 2018-11-22:
You do [set ENFPORT to 1 for the management route](https://github.com/openil/sja1105-tool/blob/master/src/lib/dynamic-config/l2-lookup.c#L153), right?

sylane, 2018-11-23:
What I was saying is that the packet was dropped **if there wasn't any management route for it**. In our prototype we are creating the management route for every packets and it works fine. And yes, we set `ENFPORT` to 1. Everything is working, I was just concern about how I would have to setup these management route with the new design...

sylane, 2018-12-03:
So we configure the MAC addresses for management trafic (not multicast in our case, but it is not important) in the configuration, we load it, then through a netdev socket for each of the switch ports and use this socket to send the packets ? Is the kernel smart enough to add a management route automatically when required ? Not sure I am understanding it well enough.

vladimiroltean, 2018-12-03:
> Will it be possible to delete/change/add vlan route entries dynamically without reloading the full configuration ? 

Yes, dynamic reconfiguration is one of the major raisons d'etre of the kmod reimplementation.
To the point of VLAN tables:
* Thanks to @gwaibel s contribution, the current pull request already supports VLAN reconfig, but for P/Q/R/S, so support for E/T - the hardware on the LS1021A-TSN board) still needs to be added
* The standard way of doing this is via switchdev operations. Currently this is done via a sysfs. It would be very much appreciated if you could add switchdev operations for FDB and VLAN manipulation - you can take the existing code both in this pull request and this one: https://github.com/openil/sja1105-tool/pull/33

> In addition we will most probably need to dynamically change TSN related configuration but I didn't get to this yet.

Dynamic reconfiguration is not hardware-supported for the time-aware scheduler unfortunately.

sylane, 2018-12-03:
We are planing on using the Q or S version of the switch fabric anyway, we will change the chip in the LS1021A box for the next prototype.

Is the dynamic reconfiguration for the time-aware scheduler not supported due to some problem, or just because it is not implemented ? Because the main reason why we need to update configuration dynamically is to not impact established TSN links. We will need to implement ieee 802.1Qcc 802.1Qat, and that means changing scheduler configuration dynamically without interfering with the already configured streams...

vladimiroltean, 2018-12-03:
> Is the dynamic reconfiguration for the time-aware scheduler not supported due to some problem, or just because it is not implemented ?

I mean the schedule-related tables are only reconfigurable via the static config table (aka "staging area"), which requires a switch reset. This is true for all of T, Q and S series switches. For TSN implementations that are more compliant to the IEEE standards (which do specify reconfiguration ability) you'd be looking at e.g. the integrated TSN switch on the LS1028.

sylane, 2018-12-03:
This is bad news for us... The LS1028 looks to much for our embedded board, and it wouldn't have enough ports; We were planing on chaining two SJA chips to get 6 switched ports.

vladimiroltean, 2018-12-03:
> too much 

If you're concerned about e.g. GPU, there will also be parts without that. But otherwise it is a dual-core A72 whereas LS1021 is A7, and there will also be a single-core version.

> wouldn't have enough ports

In terms of externally-connected ports, the LS1028 would have 4 switched ports and 2 regular MACs. You could chain the embedded L2 switches just like you could chain the SJA's.

sylane, 2018-12-03:
I am not sure I am following what you mean by chaining the switches with the LS1028, the switch fabric is integrated. If we chained it it would need to be with something like a SJA chip and then we would be back to the issue of not being able to change the TSN configuration dynamically.

vladimiroltean, 2018-12-03:
> If we chained it it would need to be with something like a SJA chip

I was actually proposing the possibility of a multi-LS1028 board.

vladimiroltean, 2018-12-03:
To be 100% clear, the only practical effect of not supporting dynamic reconfig for the TAS is that there will be some small downtime during the parameter change. How often do you expect to need to change the time schedule?

sylane, 2018-12-03:
The thing is that we will be offering an API to do TSN path reservation in a mesh network... So the frequency of the changes is out of our control, and changing the configuration of one switch could impact a lot of stream path...

closed 2019-05-06 by vladimiroltean