nxp-archive/openil_sja1105-tool #39: Loading a configuration file is not always reliable

sylane, 2018-11-29:
Trying to load the configuration file [config.error.xml](https://gist.github.com/sylane/a16826bf192c365d939f67bceb2e478e) on one of my LS1021ATSN devkit, `sja1105-tool` randomly fail:
```
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
configuration is invalid
static_config_flush failed
staging_area_flush failed
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
configuration is invalid
static_config_flush failed
staging_area_flush failed
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
configuration is invalid
static_config_flush failed
staging_area_flush failed
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
local crc failed while uploading config
static_config_flush failed
staging_area_flush failed
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
[root@etsn1:~]# /usr/bin/sja1105-tool config load -f config.error.xml 
configuration is invalid
static_config_flush failed
staging_area_flush failed
```
Rebooting the router did not change anything.
The same config load fine on my two other routers that have the exact same software installed.
The error descriptions suggest that this has nothing to do with the switch fabric, that it is a validation error, what could cause this ?

Thank you for your help.

vladimiroltean, 2018-11-29:
Sounds like either a filesystem corruption, or a memory corruption. The latter is more likely since I think we went already through the filesystem corruption possibility in a previous ticket and we ruled that out.
@Kay-Potthoff opened [this pull request](https://github.com/openil/sja1105-tool/pull/29) with some memory corruption fixes a while ago. I didn't merge it because some of the fixes were improper and I didn't have time to rework them. You can try to apply them though and see if it helps.

sylane, 2018-11-30:
```

vladimiroltean, 2018-11-30:
While running `watch -n 1 "sja1105-tool conf load -f config.error.xml; echo \$?"`, I never see any output other than 0.
Question: Is the NETCONF server running on your system? (`ps aux | grep netopeer`) If it is, can you stop it? (`/etc/init.d/S90netconf stop`)

sylane, 2018-11-30:
Netconf is not running, I had to remove it as it was making `sja1105-tool` even more unreliable.

sylane, 2018-11-30:
We build the system using OpenIL `OpenIL-201808`, is this patch supposed to be in ? Because it doesn't look like:
```
[root@etsn1:~]# ls -la /proc/device-tree/soc/dspi@2100000/spidev@1/
total 0
-r--r--r--    1 root     root             4 Jan  1 05:56 #address-cells
-r--r--r--    1 root     root             4 Jan  1 05:56 #size-cells
drwxr-xr-x    2 root     root             0 Jan  1 05:56 .
drwxr-xr-x    6 root     root             0 Jan  1 05:56 ..
-r--r--r--    1 root     root            14 Jan  1 05:56 compatible
-r--r--r--    1 root     root             7 Jan  1 05:56 name
-r--r--r--    1 root     root             4 Jan  1 05:56 reg
-r--r--r--    1 root     root             4 Jan  1 05:56 spi-max-frequency
[root@etsn1:~]# cat /etc/sja1105/sja1105.conf
[spi_setup]
	staging_area = /lib/firmware/sja1105.bin
	device       = /dev/spidev0.1
	bits         = 8
	speed        = 1000000
	delay        = 0
	cs_change    = 0
	mode         = SPI_CPHA
	dry_run      = false
	auto_flush   = false

[general]
	screen_width     = 120
	entries_per_line = 10
	verbose          = false
	debug            = false
```

sylane, 2018-11-30:
That doens't explain why I have one router without any error, one with very few and a lat one with a lot of them... Same software, same config...

vladimiroltean, 2018-11-30:
It should have went into that release tag according to https://github.com/openil/linux/commits/OpenIL-linux-201808, I don't know where you have it. When I made the patch it was because of some weirdly-behaving board during PTP commands, there seems to be some variability on the SPI interface. It would be good if we could at least rule that out.

sylane, 2018-11-30:
I tried with speed `500000` and `100000`, same errors...

sylane, 2018-11-30:
I randomly get `local crc failed while uploading config`, `global crc failed while uploading config` and `configuration is invalid`.

vladimiroltean, 2018-11-30:
Yes, I get that, those are all responses from the switch.
To prove to you that it's not the program that is sending bad data, do set `dry_run = true` in `sja1105.conf` and compare the dumps from a functional and a non-functional board using the same xml. If they're the same it's most likely an SPI issue, and you should at least try to apply the patch that repairs the timing violation between the chip select and the clock signal (which is still there at 500 and 100 KHz).

sylane, 2018-11-30:
I just tried `dru_run = true` but the output is different fro every run, the 6th and 7th bytes of the first transfer keep changing...

sylane, 2018-11-30:
Is there a way to dump the SPI without doing a dry_run ? This would be the only way to be sure what was sent on the SPI when failing.

vladimiroltean, 2018-11-30:
Ok, that's interesting, so we're back at the memory corruption hypothesis.
Could you instrument the code around [this area](https://github.com/openil/sja1105-tool/blob/master/src/tool/staging-area.c#L291)? Obviously `sja1105_static_config_check_valid` didn't trigger a validation error at the point it was called, perhaps call it again after `static_config_upload` (SPI transfer)?

> Is there a way to dump the SPI without doing a dry_run ?

Only using a probe, or dumps in the spidev driver, I think. But there are some statistics under `ls -la /sys/bus/spi/devices/spi0.1/statistics/` that you can try to check out.

sylane, 2018-11-30:
I don't thing the changes in the first SPI transaction is relevant, because it happens all the time on all my routers. It sounds more like some time or random related configuration...

sylane, 2018-11-30:
I can confirm that on all my routers, beside the first transaction bytes 6th and 7th, the SPI transactions in dry run are identical for all runs.

vladimiroltean, 2018-11-30:
First command (`(0000): 80 00 01 10  58 9C 9E BF`) is a buggy `sja1105_inhibit_tx` (`ETH_PORT_CONTROL_ADDR = 0x11`) - buggy because there's no memset of the buffer to zero - that's the difference we're looking at).
Second command is a reset (`(0000): 81 00 44 00  00 00 00 04`).

sylane, 2018-11-30:
The data generated on the router that fail often and the router that almost never fail is the same (beside the first buggy transaction). So this is definitely something related to the switch state.

vladimiroltean, 2018-11-30:
You can try to insert delays between the `sja1105_cold_reset` operation and the `static_config_upload` if you believe it to be because the switch is in a state where it doesn't accept a new static config yet.

sylane, 2018-11-30:
It fails even at speed `10000` so this doesn't seem related to speed.

vladimiroltean, 2018-11-30:
Could you try to apply the SPI timing violation patch first, before we examine more exotic possibilities?

sylane, 2018-11-30:
Are you talking about this : https://github.com/openil/linux/commit/d23888f9b3e07be00b77ebb61bd70175f8b42e17 ?
If it is I am not sure how to apply it. I just built the image using OpenIL, I am no expert on buildroot.

vladimiroltean, 2018-11-30:
Just go to your buildroot tree, then `cd output/build/linux-*`, then `patch -p1 < file.patch`, then `rm .stamp_built`, then go back to the buildroot top dir and run `make linux` to produce an updated kernel image.

sylane, 2018-11-30:
I applied the patch, removed `.stamp_built` called `make linux` then later on `make` (part of our makefile)... With the new image it seems the patch is not there:
```
[root@etsn1:~]# ls -la /proc/device-tree/soc/dspi@2100000/spidev@1/
total 0
-r--r--r--    1 root     root             4 Jan  1 00:04 #address-cells
-r--r--r--    1 root     root             4 Jan  1 00:04 #size-cells
drwxr-xr-x    2 root     root             0 Jan  1 00:04 .
drwxr-xr-x    6 root     root             0 Jan  1 00:04 ..
-r--r--r--    1 root     root            14 Jan  1 00:04 compatible
-r--r--r--    1 root     root             7 Jan  1 00:04 name
-r--r--r--    1 root     root             4 Jan  1 00:04 reg
-r--r--r--    1 root     root             4 Jan  1 00:04 spi-max-frequency
```
Not sure what I did wrong...

sylane, 2018-12-03:
I'll test it. I would say it should work given that I am now calling the tool multiple times if it fails and as a result the prototype seems to work.

vladimiroltean, 2018-12-03:
If it works it points to an unreliable communication channel over SPI. To confirm this is the case, I would suggest to run `sja1105-tool reg 0` (read the device ID) in a loop overnight and record the number of times you get an output different than `0x00000000: 9e00030e`.

sylane, 2018-12-03:
I tested the patch. It seems to work, though it would be nice to log that the operation was successful otherwise it sounds like it failed when retrying:
```
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
local crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
...
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
configuration is invalid
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
...
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
local crc failed while uploading config
local crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
...
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
global crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
...
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
local crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
...
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
configuration is invalid
local crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
global crc failed while uploading config
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
0
[root@etsn1:~]# sja1105-tool config load -f config.error.xml; echo $?
global crc failed while uploading config
0
```

vladimiroltean, 2018-12-03:
> it would be nice to log that the operation was successful

I don't intend to merge this patch yet, as I'm not convinced that it's the correct fix for the problem you're having. It would be good if you could perform some integrity checking on your SPI link.

sylane, 2018-12-03:
I'll run it over night to be sure, but for now I run it ~5k times without any error (talking about `sja1105-tool reg 0`).

sylane, 2018-12-05:
So I run `sja1105-tool reg 0` for a very long time and it always read the proper value. Any more idea on how to check the integrity of the SPI link ?