nxp-archive/openil_sja1105-tool #41: Is the ports' status related to the PHY or is it specififc to the SJA chip ?

sylane, 2019-01-07:
The ports' status information is specific of the SJA chip or does it mirror in some ways the PHY hardware of the port ? And if it is specific, who could I get the information ? Is it even possible ?

vladimiroltean, 2019-01-07:
Per the [known issues](https://github.com/openil/sja1105-tool#known-issues):

>The 5 ports of the SJA1105 switch do not have a Linux kernel driver. As such, the PHY chip (BCM5464R) attached to the 4 externally connected ports of the LS1021ATSN board (ETH2, ETH3, ETH4, ETH5) is not controlled by the Linux kernel either.

A proper solution to this is given in the kmod pull request (still unmerged): https://github.com/openil/sja1105-tool/pull/36:

> Dynamically adjust a port's MAC speed on link status change.

sylane, 2019-01-07:
With the current tool it seems reading the registers `PORT_STATUS_MII[0-4]` is mirroring the PHY info, would that be correct ?

vladimiroltean, 2019-01-07:
The SJA1105 does not have management connection (MDIO) to the external BCM5464R quad RGMII PHY. Therefore, PORT_STATUS_MII[0-4] is not mirroring the external PHY settings.
My hunch is that those registers simply inform the user of the settings in the MAC Configuration Table, since both the static configuration table, and the dynamic reconfiguration entries for the MAC are write-only.

closed 2019-01-11 by sylane