nxp-archive/openil_sja1105-tool #42: TSN with PTP clock source sometime is just not working

sylane, 2019-01-11:
I am testing TSN with two routers. When setting the clock source to PTP with setting `clksrc` in `schedule-entry-points-parameters-table` to `3` it just seem to not do any scheduling. I setup the scheduler to enforce a priority to have a constant and guaranteed latency of 1 ms and it is just not enforced with the clock source being PTP, if I set it to `1` (standalone mode) it behave as expected, with one router the packet is guaranteed 1 ms latency, but then with two routers it is ~ 2 ms because they are not synchronized...
Yesterday PTP clock wasn't working either but after some time it seemed to start working, and today it doesn't work again.

sylane, 2019-01-11:
So I probably misunderstood wht this offset was. I though it was the maximum difference between the sja clock and the ptp clock... So I set a large value because I want the sja clock to always be updated.

vladimiroltean, 2019-01-11:
Can you share a log of ptp4l with `-m -l 7`? Your understanding of offset and `-t` option is correct.

sylane, 2019-01-11:
Here are the condole for the 3 routers:

sylane, 2019-01-11:
I see a lot of `sja1105: sync timer timeout` and on etsn1 I see a lot of `port 1: master sync timeout` too.

vladimiroltean, 2019-01-11:
I'm afraid you're simply dealing with plain old integer overflow. The timeout argument (`-t`) is internally multiplied by 1000. The if condition I linked you to will never start the Qbv engine.

vladimiroltean, 2019-01-11:
Board etsn1 is acting super weird. I've heard about rare instances of such malfunction (hence the `sja1105_ptp_clk_get: returned zero` print in the code), but never actually had a board behave like that in front of me. Try to send it back if you can. If not, I'll try to find some time and ask you to do some tests on it.

sylane, 2019-01-11:
ok, it makes sense. What is the use of this offset ? In what case I wouldn't want the SJA clock to be updated from the PTP clock ? Isn't it possible that the first time the clock is updated the offset could be very large ? I am fin with setting a smaller offset if I could be sure they will not just stop synchronizing because the difference is too large...

sylane, 2019-01-11:
ok, using `-t 200` I got qvb with PTP clock source working again. It is weird yesterday it worked for a time with `-t 2000000`...

vladimiroltean, 2019-01-11:
The `timeout` messages are not indicative of a problem. See this thread for more details: https://github.com/openil/openil/issues/6

vladimiroltean, 2019-01-11:
The `-t` option is for resetting the SJA1105 clock if the offset exceeds the limit. In truth that shouldn't be done, but instead just map the clock as a standard Linux clock and let ptp4l do its thing with a proper servo (which is clock stepping initially, then frequency adjustment). The kmod implementation tries to do just that. Still not perfectly calibrated, but the code to support scheduling with PTP time source is already in the pull request if you want to try it out.

sylane, 2019-01-11:
Now the scheduling is working but it seems to me the router are not really synchronized. I did a schedule with a large cycles similar to the demo:
  - router 1: `|8ms open|1ms guard|1ms output|8ms open|1ms guard|1ms input|`
  - router 2: `|8ms open|1ms guard|1ms input|8ms open|1ms guard|1ms output|`
Without any scheduling the roundtrip time is around 100us, so in 1ms there should be way enough time for the packet to go through the two routers and back. If the routers where synchronized and everything was correct the RTT should be very close around 20ms but I get one very close to 40ms...

This clearly look like the router are not synchronized, how could I check/debug these kind of issues ?

sylane, 2019-01-11:
I restarted ptp4l and now it give me a constant 20ms... weird...

sylane, 2019-01-11:
Given I don't have the nodejs tool used to measure the latency in the demo, I am using `ping -A` directly. I am a bit confused about the description of PIT in the demo, it says it measure the time between ping requests on the RECEIVER, how do you do that ? For my measure I either use the RTT given by `ping` or use `ping -D` and use the the timestamp difference:

  - RTT: `sudo ping -c $COUNT -DA $ADDR | tail -n +2 | head -n -4  | sed -e 's/\[\([.0-9]*\)\].*time=\([.0-9]*\) ms/\1 \2/' | awk "{if(length(ref)==0) ref=\$1}{print (\$1-ref)*1000, \$2}"`
  - PIT: `sudo ping -c $COUNT -DA $ADDR | tail -n +2 | head -n -4  | sed -e 's/\[\([.0-9]*\)\].*time=\([.0-9]*\) ms/\1 \2/' | awk "{if(length(ref)==0) ref=\$1}{val=(\$1-last)*1000}{if(val<$MAX_FILTER) print (\$1-ref)*1000, val}{last=\$1}"`

Is what I am measuring really PIT ? I don't really see how it is better than the RTT, at least the way I am doing it...

vladimiroltean, 2019-01-15:
In the ptp4l output, there are 2 sets of offsets you need to watch out for: the master-to-slave offset (goes over Ethernet) and the slave-to-local-slave (SJA1105 synchronized over SPI).
The demo web app really measures packet inter-arrival time. It is putting one board in promiscuous mode and recording the software timestamps from when the packet reaches tcpdump: https://github.com/openil/openil/blob/master/package/nodejs-lbt/src/server.js#L200 The delta between consecutive packets is done with `tcpdump -ttt`.

closed 2019-05-06 by vladimiroltean