nxp-archive/openil_sja1105-tool #43: Some confusion about the SJA TSN configuration

sylane, 2019-01-14:
I was able to test the basic TSN scenario described in the OpenIL user guide, but trying to understand the remaining of the TSN-related configuration of the SJA chip is a challenge. In particular, I want to understand how to configure multiple TSN subschedules with multiple entry points. The documentation is a bit obscure, so I wanted to ask some questions and present some assumptions I am not quite sure about that may need clarifications. If anybody could help to shed light on any of these points it would be fantastic. Note that I am using the SJA1105T, there may be differences with other chips.

Are these correct assumptions?

 1. At most 8 subschedules can be defined.
 2. The `actsubsch` field of `schedule-entry-points-parameters-table` must be set to the number of subschedules minus one; if the subschedule 0, 3 and 5 are active the value of `actsubsch` would be `2` (???).
 3. Table `schedule-entry-points-table` defines the subschedules running concurrently.
 4. The same subschedule can be scheduled multiple times with different `delta` values.
 5. The index of the first entry in `schedule-table` of a subschedule is defined in the `address` field of a `schedule-entry-points-table` entry.
 6. The index of the last entry of the subschedule is defined the `subscheind` table in `schedule-parameters-table`, the index in the table itself being defined by the `subschindx` filed in a `schedule-entry-points-table` entry.
 7. The fields `winstindex`, `winst` and `winend` define a time-triggered window where input packets of a stream defined in the `VL` tables are accepted. Out of this window, the input packets are dropped (???).

Some questions:

 8. The documentation (UM10944 4.2.2) state that for each entry in the `schedule-table` there should be 8 entries in `schedule-entry-points-table`, but the demo in OpenIL user guide itself only set a single entry point for a single subschedule of 6 entries. That would explain why the `schedule-entry-points-table` can have 2048 entries, but I don't understand why each subschedule entries would require its own entry point... What am I missing?
 9. What is the use-case (if I understood that correctly) to have multiple entry points for the same subschedule with different `delta`?
 10. I don't understand why the first and last entries of a subschedule are defined in different places, one in the entry point and the other in the parameters; what is the use-case for that?
 11. Maybe not directly related, but how the entries in `vl-lookup-table` are linked to the ones in `vl-policing-table` as stated by the documentation (UM10944 4.2.3)? I don't see any fields that could be used to associate entries between the two tables.

Thank you for reading this, and for any clarification you can give.

vladimiroltean, 2019-01-14:
I have tried in the attached picture to come up with as clear a description as I could.
The example is slightly more complex as what is in the OpenIL manual, as there are 2 concurrent subschedules (cycles) here. The terminology from the user manual is extremely confusing, with the "schedule" and "subschedule" terms being used somewhat arbitrarily, hence my preference to call them timeslots and cycles respectively, for clarity.

[image]

There seem to be more degrees of freedom in configuring the hardware than the `scheduler-create` helper script permits, however I am not aware that any practical use case someone might have is missed if just using the script and a JSON description.

1. If by "subschedules" you mean individual cycles that the scheduling engine executes in parallel by iterating with a k through the schedule-table, then yes.
2. Yes
3. Yes
4. A "subschedule" is defined by a single entry point (initial k) and delta.
5. Yes
6. Yes
7. I don't know
8. The explanation in UM10944 chapter 4.2.2 (`Each entry point into the schedule consists of 8 entries in the Schedule Entry Points table.`) is rather unfortunate.
9. It is not possible to do so
10. I don't know of any use case of separating the start and stop indices
11. Association is done by entry index (vl-policing-table[i] and vl-lookup-table[i] refer to same virtual link). See a bit more details here https://github.com/openil/openil/issues/11#issuecomment-420707589

sylane, 2019-01-15:
Thank you very much your insights ! It is very helpful.
This response should really stick there as it could help a lot of people to better understand the documentation.

Two more questions if it is not taking too much advantage of your time:

 1. If the only two active cycles were to be cycle `0` and cycle `5`, would `actsubsch` be `1` (the total number of cycles minus 1) or `5` (the maximum value of the `address` field of all the entry points) ?
 2. If there is a maximum of 8 cycles defined and there is at most an entry point per cycles, why would the `schedule-entry-points-table` table allow up to 2048 entries ?

Thank you very much.

vladimiroltean, 2019-01-15:
1. The `actsubsch` should be 1.
2. You're probably talking about this phrase:
> The entry point must contain exactly one valid entry for each active subschedule (the table thus provides space for up to 2048 entries).

I don't know why it's 2048 entries in size. I'm not even completely sure it is that big in reality. I'm forwarding this question internally, but I would just disregard the fact that it's documented as 2048 and just assume it's an 8 entry sized table.

mirceachimerel, 2019-01-16:
More details: There are 3 topics: the clock synchronization topic (SAE AS6802), the time-triggered traffic topic, the input policing topic. 

a) The clock synchronization topic:
The clock synchronization of the network is realized with protocol control frames (a.k.a. PCF) which are generated periodically by the clock synchronization masters with a configurable period called integration cycle period. A PCF frame has a field named integration cycle, which is an ID (or number) of the PCF, in range 0 to 255. The PCF frames should be generated by the clock synchronization masters using consecutive integration cycle IDs from value 0 to MAXINTEGCY (UM10944 4.2.16), where MAXINTEGCY is in range 0 to 255. After the PCF frame with integration cycle ID MAXINTEGCY is generated by the clock synchronization masters, the next PCF frame generated by the the clock synchronization masters must have integration cycle ID set to 0. And so on. 
A synchronization cluster consists of all integration cycles having different IDs. The synchronization cluster starts at the beginning of integration cycle ID 0 and ends at the end of the integration cycle ID MAXINTEGCY.

b) The time-triggered traffic topic:
The time-triggered traffic can be relayed only when the network is synchronized using PCF frames. When the network is decided to be stable (SAE AS6802), the schedule is started, if configured. The starting point in time of the schedule can happen at the beginning of any integration cycle. The schedule can start in integration cycle ID 0, or in integration cycle ID 1, or in integration cycle ID 2, ... , or in integration cycle ID MAXINTEGCY. If MAXINTEGCY is set to 255, there are 256 starting points of the schedule because there are 256 integration cycles.    
The schedule-table of SJA1105T has 2048 entries which can be allocated to up to 8 subschedules. The number of active subschedules is set with parameter ACTSUBSCH, in range 0 to 7.     
There can be only one subschedule with 2048 entries or less. 
What is the schedule and what is the subschedule? 
The schedule is the list of events scheduled in the schedule table (UM10944 4.2.1) which is executed continuously using up to 8 execution threads called subschedules. The schedule table contain action events for reception or transmission of frames. These threads (subschedules) execute entries in paralel, one by one, from the schedule table (UM10944 4.2.1). There are no gaps is the schedule table between the entries executed by the active subschedules. The schedule table can be accessed only by one thread (subschedule) at a particular point in time, in order to execute the entry. The execution time of an entry is given by parameter DELTA from (UM10944 4.2.2) when the schedule execution is started and the threads (subschedules) start to execute, or by parameter DELTA from (UM10944 4.2.1) when the threads (subschedules) are running.  

Lets consider as an example the following schedule configuration: MAXINTEGCY (UM10944 4.2.16) set to 4, 4 active subschedules with IDs 0, 3, 4, and 7, and subschedule ID 0 having 14 entries, subschedule ID 3 having 3 entries, subschedule ID 4 having 20 entries, subschedule ID 7 having 100 entries. 
The schedule parameters/table should be set as follows: 
(UM10944 4.2.16) MAXINTEGCY = 4
(UM10944 4.2.12) ACTSUBSCH = 3, 
(UM10944 4.2.11) SUBSCHEIND(0) = 13, SUBSCHEIND(1) = 13, SUBSCHEIND(2) = 13, SUBSCHEIND(3) = 16, SUBSCHEIND(4) = 36, SUBSCHEIND(5) = 36, SUBSCHEIND(6) = 36, SUBSCHEIND(7) = 136. 
(UM10944 4.2.1)  The entries 0 to 136 must be configured in schedule table. When the schedule is running, the entries 0 to 13 are executed continuously by subschedule (thread) ID 0, the entries 14 to 16 are executed continuously by subschedule (thread) ID 3, the entries 17 to 36 are executed continuously by subschedule (thread) ID 4, the entries 37 to 136 are executed continuously by subschedule (thread) ID 7.
The schedule-entry-points-table (UM10944 4.2.2) has 2048 entries and it has nothing to do with the size of 2048 entries of the schedule table (UM10944 4.2.1).
The size of table schedule-entry-points-table (UM10944 4.2.2) is the maximum number of integration cycles (256) (see topic a)) multiplied by the maximum number of subschedules (execution threads) (8) resulting a number of 2048 entries. In the example above, MAXINTEGCY (UM10944 4.2.16) is set to 4 meaning that in schedule-entry-points-table must be configured 5*8=40 entries (0 to 39). The entries 0 to 7 are allocated to integration cycle ID 0, the entries 8 to 15 are allocated to integration cycle ID 1, the entries 16 to 23 are allocated to integration cycle ID 2, the entries 24 to 31 are allocated to integration cycle ID 3, the entries 32 to 39 are allocated to integration cycle ID 4. If the schedule is started in integration cycle ID 3, the initial values (DELTA and ADDRESS from (UM10944 4.2.2)) are used to initialize the active subschedules. 
In our example, only schedule-entry-points-table entries (UM10944 4.2.2) 24, 25, 26 and 27 are used because (UM10944 4.2.12) ACTSUBSCH is 3. 
Entry 24 of schedule-entry-points-table (UM10944 4.2.2) must have SUBSCHINDX = 0, DELTA = i, ADDRESS = a0 in range 0 to 13.  
Entry 25 of schedule-entry-points-table (UM10944 4.2.2) must have SUBSCHINDX = 3, DELTA = j, ADDRESS = a1 in range 14 to 16.  
Entry 26 of schedule-entry-points-table (UM10944 4.2.2) must have SUBSCHINDX = 4, DELTA = k, ADDRESS = a2 in range 17 to 36.  
[... cut ...]
If a VL is configured as time-triggered VL, a received event for that VL must be scheduled in the schedule table (UM10944 4.2.1), otherwise the frame is dropped. The receive event is scheduled with parameters WINST, WINEND and WINSTINDEX (UM10944 4.2.1). 
When a entry in the schedule table (UM10944 4.2.1) is executed by the subschedule and WINST is set to 1, a receive window is opened for the table entry (UM10944 4.2.4) pointed by parameter WINSTINDEX.      
When a entry (different from previous one) in the schedule table (UM10944 4.2.1) is executed by the subschedule and WINEND is set to 1, a receive window is closed for the table entry (UM10944 4.2.4) pointed by parameter WINSTINDEX.

sylane, 2019-01-22:
Thank you very much @mirceachimerel, this is very informative. But I am now more confused about the integration cycle... Is this something generated by the PTP grandmaster and forwarded to the switch by the patched ptp4l service ? What is the use-case for having MAXINTEGCY > 0 and multiple entry point for the same sub-schedules at different integration cycle ? Is it only so the switch synchronize faster, that it doesn't have to wait for the integration cycle ID to overflow ?

mirceachimerel, 2019-01-22:
_"But I am now more confused about the integration cycle... Is this something generated by the PTP grandmaster and forwarded to the switch by the patched ptp4l service ? "_

The "integration cycle" is specific to SAE AS6802, selected with CLKSRC = 10 (UM10944 4.2.12). It is the interval between the 2 consecutive PCFs that are used to synchronize the network. It has nothing to do with PTP grand master. 
For PTP synchronization, MAXINTEGCY is irrelevant, considered 0. In this case, everything related to schedule and subschedules (execution threads) (from the previous message) remain valid with the constraints that in schedule-entry-points-table (UM10944 4.2.2) only 8 entries are needed to be configured because MAXINTEGCY is considered 0. But, you need to configure (UM10944 4.2.12) ACTSUBSCH, (UM10944 4.2.11) SUBSCHEIND(i), schedule table (UM10944 4.2.1). Also, it is possible to configure receive triggers in the schedule table (UM10944 4.2.1) if the sending nodes can transmit traffic synchronized with the subschedules (execution threads).     

_"What is the use-case for having MAXINTEGCY > 0 and multiple entry point for the same sub-schedules at different integration cycle ? Is it only so the switch synchronize faster, that it doesn't have to wait for the integration cycle ID to overflow ?"_

Yes.

RikusW, 2019-03-04:
Will there be an Application hint released regarding Qbv configuration sometime ?

vladimiroltean, 2019-03-04:
What is being discussed in this ticket is not IEEE 802.1Qbv, it is SAE AS6802 (aka Time-Triggered Ethernet).
It is possible to configure this time-aware shaper so as to behave more-or-less compatibly with IEEE 802.1Qbv, and use IEEE 1588 time reference. If this is what you want to achieve, I don't believe that anything in this thread will make you any less confused on how to achieve that. In that case I would recommend starting off with one of the [documented examples](https://github.com/openil/sja1105-tool/blob/master/src/helpers/configs/synchronized-qbv/qbv-ptp.sh#L64) and reading the OpenIL manual for SJA1105. I would not advise trying to understand the low-level internals of the `scheduler-create` script, since it configures only a very small subset of the TTEthernet features.

RikusW, 2019-03-04:
I'm already running a custom 802.1AS PTP in bridge mode with Qbv startup code on the LS1021ATSN box. Was just wondering about the scheduler configuration.

vladimiroltean, 2019-03-04:
Don't know if you're aware or not, but chapter 7.10 of https://www.openil.org/news/files/OpenIL_User_Guide_Rev1.3.pdf documents all that is supported with Qbv and PTP on the SJA1105.

closed 2019-05-06 by vladimiroltean