nxp-archive/openil_sja1105-tool #45: Qbv scheduler documentation

RikusW, 2019-03-04:
Is there any documentation regarding the scheduler configuration available ? Even if we have so sign a NDA ? I work for a company specializing in AVB and TSN equipment, https://tsn.systems/en/about-tsn-systems/
I already have UM10944.pdf and used that, the patched linuxptp source and some tickets here on github to get 802.1AS with Qbv startup working on the SJA1105.
Thanks

vladimiroltean, 2019-03-04:
Hi Rikus,
All documentation available for configuring the SJA1105 is in the [OpenIL manual](https://www.openil.org/news/files/OpenIL_User_Guide_Rev1.3.pdf).
For the new LS1028, the configuration model is different, and will also be documented in OpenIL.
There is no other documentation for configuring TSN on OpenIL-supported hardware, even under NDA, that I can link you to.
Please let me know if there is any particular question I can help address.

RikusW, 2019-03-04:
Hi Vladimir
How much does the SJA1105 scheduler differ from 802.1Qbv ? I already saw the comment regarding reset required for reloading the schedule.

vladimiroltean, 2019-03-04:
Practical differences that I am aware of:
* It does not support anything that is equivalent of ADMIN parameters and OPER parameters. There is only one set of configuration parameters (equivalent to OPER) at any given moment in time.
* Time-aware scheduler configuration (gate control list) is loaded via a static configuration, and therefore cannot be changed at runtime without temporarily disrupting traffic (aka switch reset). But setting activation time in the future is supported.
* If the scheduler is at the end of a time window and transmitting a frame would take longer than the time left in this open window, IEEE requires to either not transmit the frame, or increment a transmission overrun counter. SJA1105 does neither, and transmits the frame regardless. It depends whether this is a problem or not - there are complications in implementation that follow IEEE more strictly as well: if a frame is longer than its maximum time window, it may never get transmitted at all, and due to head-of-line blocking, the whole transmission queue will be forever blocked.

RikusW, 2019-03-07:
Does the SJA1105 support per port scheduling ?

vladimiroltean, 2019-03-07:
Could you please ask in more detail?

RikusW, 2019-03-07:
It seems possible to run a subschedule for every port by using Schedule table DESTPORTS set to only one port. My only concern is that it doesn't seem possible to start them all at the same time due to the DELTA value in the Schedule Entry Points table.

Table 4. Schedule Entry Points table (block 01h) (From UM10944.pdf)
DELTA This field defines the delay before this entry fires in multiples of 200 ns. Any two entries of an
entry point assigned to an active subschedule must contain **different** values for this field. A
value of zero is not allowed. An arbitrary value may be provided for inactive subschedules.

vladimiroltean, 2019-03-07:
I don't think you would want to run subschedules (cycles) in parallel. If you do that, indeed you make sure that no time window opens at the same time in two subschedules, neither now nor ever in the future. There is no software logic in place currently to detect that condition.
But I think a single cycle should do the job just fine. Cycles are not per-port. Each gate control list entry from within a cycle (or subschedule, which way you want to call it) has a port list on which the specified open/close action should be run.
Do you specifically require to run different cycles in parallel on different ports? What use do you have for that?

RikusW, 2019-03-07:
From the 802.1Qbv spec -> "A gate control list associated with each Port contains an ordered list of gate operations."   A "gate control list" would be equivalent to a cycle.

vladimiroltean, 2019-03-07:
Well, then your understanding is correct and this is another deviation from what IEEE specifies.

vladimiroltean, 2019-03-07:
What are you trying to do with the switch time scheduler?

RikusW, 2019-03-08:
The end goal is to have a TSN compliant switch, as it is it will work for some use cases. Having a cycle set to address a single port will be equivalent to a Qbv gate control list. My main concern is whether two or more cycles, each only set to update a single port, can update the gates in the same time period ? It seems entirely possible to do this 200ns apart but that is not ideal.

RikusW, 2019-03-08:
I'm not quite sure how PTPCLKCORP affects the scheduler, I've seen the patched linuxptp setting it to the cycle length, having a block diagram of how the whole scheduler works would be quite helpful.

RikusW, 2019-03-08:
The other issue I had was getting PTPTSCLK and PTPCLKVAL at the exact same time. Hardware support will be required to latch both on reading, I got a jitter of +-2us when reading it in software using the same method as in the patched linuxptp. I ended up using CORRCLK4TS and a servo to get a grandmaster offset of +-100ns.

vladimiroltean, 2019-03-08:
> getting PTPTSCLK and PTPCLKVAL at the exact same time

A small optimization I can probably do is expose an API for reading both at the same time.
The function would perform a single SPI read operation ranging from address 0x18 (PTPCLKVAL) to 0x1C (PTPTSCLK), and retrieve both from the same SPI buffer.

RikusW, 2019-03-08:
I actually did exactly this, it did help somewhat with jitter, but then there will be 96 SPI bit clock periods introduced so the SPI clock will influence the PTPCLKVAL offset and jitter.

void sja1105_read_clks(uint64_t *tsclk, uint64_t *clkval)
{
	uint32_t buf[5];

	sja1105_spi_send_packed_buf(&ss, SPI_READ, 0x18, buf, 20);

	*clkval = be32toh(buf[1]);
	*clkval <<= 32;
	*clkval |= be32toh(buf[0]);
	*clkval << 3;

	*tsclk = be32toh(buf[4]);
	*tsclk <<= 32;
	*tsclk |= be32toh(buf[3]);
	*tsclk << 3;

	tsclk -= (1000000000 / ss.speed) * 96; //XXX clock correction XXX
}
+

vladimiroltean, 2019-03-08:
> I'm not quite sure how PTPCLKCORP affects the scheduler
It's quite complicated, let me gather some info from an old conversation with the switch IP vendor and I'll try to structure a response.

vladimiroltean, 2019-03-08:
> there will be 96 SPI bit clock periods introduced so the SPI clock will influence the PTPCLKVAL offset and jitter

Exactly, there will be 96 SPI bit times offset (that you can adjust for), but why jitter?

RikusW, 2019-03-08:
That will depend on the SPI clock source, rate ratios between clocks can make life quite interesting, I've seen temperature changes affect the ratios quite a lot. I thought reading it like this would improve the PTP sync but it didn't help as much as I expected it to.

                                  PTPTSCLK       PTPCLKVAL       GMTIME               (PTPCLKVAL - GMTIME)

SJA1105 CLKVAL ts 9240448436 cv 1103015400929  1103015398361    -2568 -
SJA1105 CLKVAL ts 9257231635 cv 1103032183548  1103032181005    -2543 -
SJA1105 CLKVAL ts 9272921501 cv 1103047872301  1103047870446    -1855 -
SJA1105 CLKVAL ts 9289893876 cv 1103064844867  1103064845072      205 -
SJA1105 CLKVAL ts 9305620724 cv 1103080571408  1103080571499       91 -
SJA1105 CLKVAL ts 9322103804 cv 1103097054022  1103097054069       47 -
SJA1105 CLKVAL ts 9338877425 cv 1103113826988  1103113827228      240 -
SJA1105 CLKVAL ts 9355510118 cv 1103130458470  1103130459455      985 -
SJA1105 CLKVAL ts 9371985950 cv 1103146933879  1103146934824      945 -
SJA1105 CLKVAL ts 9388612293 cv 1103163559679  1103163560703     1024 -
SJA1105 CLKVAL ts 9404343074 cv 1103179290798  1103179288130    -2668 -
SJA1105 CLKVAL ts 9421126520 cv 1103196073880  1103196071111    -2769 -
SJA1105 CLKVAL ts 9438047536 cv 1103212994332  1103212991616    -2716 -
SJA1105 CLKVAL ts 9454826799 cv 1103229773066  1103229770405    -2661 -
SJA1105 CLKVAL ts 9471448731 cv 1103246394603  1103246391870    -2733 -
SJA1105 CLKVAL ts 9488230895 cv 1103263176434  1103263173561    -2873 -
SJA1105 CLKVAL ts 9504719173 cv 1103279664065  1103279661399    -2666 -
SJA1105 CLKVAL ts 9521348990 cv 1103296293395  1103296290770    -2625 -
SJA1105 CLKVAL ts 9537231000 cv 1103312174797  1103312172351    -2446 -
SJA1105 CLKVAL ts 9553847906 cv 1103328791405  1103328788790    -2615 -
SJA1105 CLKVAL ts 9570787739 cv 1103345730551  1103345728119    -2432 -
SJA1105 CLKVAL ts 9587560276 cv 1103362502639  1103362500189    -2450 -
SJA1105 CLKVAL ts 9604342633 cv 1103379284597  1103379282077    -2520 -
SJA1105 CLKVAL ts 9620666361 cv 1103395607775  1103395605341    -2434 -
SJA1105 CLKVAL ts 9637182763 cv 1103412123696  1103412123756       60 -
SJA1105 CLKVAL ts 9653321771 cv 1103428262186  1103428262338      152 -
SJA1105 CLKVAL ts 9669954276 cv 1103444894495  1103444894303     -192 -
SJA1105 CLKVAL ts 9686014501 cv 1103460954043  1103460954131       88 -
SJA1105 CLKVAL ts 9702230060 cv 1103477169061  1103477169237      176 -
SJA1105 CLKVAL ts 9719002678 cv 1103493941293  1103493941312       19 -
SJA1105 CLKVAL ts 9735785655 cv 1103510722986  1103510723819      833 -
SJA1105 CLKVAL ts 9751561173 cv 1103526498884  1103526495800    -3084 -
SJA1105 CLKVAL ts 9768566969 cv 1103543504364  1103543501040    -3324 -
SJA1105 CLKVAL ts 9784769376 cv 1103559706185  1103559703014    -3171 -
SJA1105 CLKVAL ts 9801554130 cv 1103576490394  1103576487294    -3100 -
SJA1105 CLKVAL ts 9818170803 cv 1103593106588  1103593103494    -3094 -
SJA1105 CLKVAL ts 9834951979 cv 1103609887200  1103609884121    -3079 -
SJA1105 CLKVAL ts 9851730923 cv 1103626665823  1103626662592    -3231 -
SJA1105 CLKVAL ts 9867324083 cv 1103642258375  1103642255421    -2954 -
SJA1105 CLKVAL ts 9883654718 cv 1103658589178  1103658585562    -3616 -
SJA1105 CLKVAL ts 9900261953 cv 1103675194794  1103675192328    -2466 -

jitter is the difference between the deltas  (ts - tsold) - (cv - cvold)
                                  PTPTSCLK       PTPCLKVAL
SJA1105 CLKVAL ts 359678683 cv 359661341 jitter 1576
SJA1105 CLKVAL ts 376314797 cv 376299076 jitter -1621
SJA1105 CLKVAL ts 394786594 cv 394770467 jitter 406
SJA1105 CLKVAL ts 409032008 cv 409015733 jitter 148
SJA1105 CLKVAL ts 425364383 cv 425347627 jitter 481
SJA1105 CLKVAL ts 442169629 cv 442153551 jitter -678
SJA1105 CLKVAL ts 458832086 cv 458815856 jitter 152
SJA1105 CLKVAL ts 475637669 cv 475621417 jitter 22
SJA1105 CLKVAL ts 492462468 cv 492446309 jitter -93
SJA1105 CLKVAL ts 509239429 cv 509223201 jitter 69
SJA1105 CLKVAL ts 524978686 cv 524962411 jitter 47
SJA1105 CLKVAL ts 541753283 cv 541736920 jitter 88
SJA1105 CLKVAL ts 558116389 cv 558100642 jitter -616
SJA1105 CLKVAL ts 574758939 cv 574742766 jitter 426
SJA1105 CLKVAL ts 591702422 cv 591685235 jitter 1014
SJA1105 CLKVAL ts 608554665 cv 608538440 jitter -962
SJA1105 CLKVAL ts 625207341 cv 625190064 jitter 1052
SJA1105 CLKVAL ts 640922545 cv 640906244 jitter -976
SJA1105 CLKVAL ts 657705000 cv 657687709 jitter 990
SJA1105 CLKVAL ts 674046640 cv 674030419 jitter -1070
SJA1105 CLKVAL ts 690680765 cv 690664438 jitter 106
SJA1105 CLKVAL ts 707723280 cv 707707544 jitter -591
SJA1105 CLKVAL ts 724802804 cv 724786580 jitter 488
SJA1105 CLKVAL ts 740986998 cv 740971169 jitter -395
SJA1105 CLKVAL ts 757637593 cv 757621306 jitter 458
SJA1105 CLKVAL ts 773630140 cv 773613834 jitter 19
SJA1105 CLKVAL ts 790274795 cv 790258583 jitter -94
SJA1105 CLKVAL ts 806747231 cv 806730911 jitter 108
SJA1105 CLKVAL ts 823539347 cv 823522055 jitter 972
SJA1105 CLKVAL ts 839290208 cv 839274235 jitter -1319
SJA1105 CLKVAL ts 855374913 cv 855358616 jitter 324
SJA1105 CLKVAL ts 872167279 cv 872151012 jitter -30
SJA1105 CLKVAL ts 888908725 cv 888891688 jitter 770
SJA1105 CLKVAL ts 905084052 cv 905067820 jitter -805
SJA1105 CLKVAL ts 921558191 cv 921542080 jitter -121
SJA1105 CLKVAL ts 938337306 cv 938320909 jitter 286
SJA1105 CLKVAL ts 954541173 cv 954522763 jitter 2013
SJA1105 CLKVAL ts 970623020 cv 970607354 jitter -2744
SJA1105 CLKVAL ts 987275358 cv 987259073 jitter 619

vladimiroltean, 2019-03-08:
The schedule engine and the PTP clock are driven by the same oscillator, and they run in parallel. But whilst the PTP clock can keep an absolute time-of-day, the schedule engine is only running in 'ticks' (25 ticks make up a delta, which is 200ns), and wrapping around at the end of each cycle.
The schedule engine is started when the PTP clock reaches the PTPSCHTM time (in PTP domain).
Because the PTP clock can be rate-corrected (accelerated or slowed down) by a software servo, and the schedule engine clock runs in parallel to the PTP clock, there is logic internal to the switch that periodically keeps the schedule engine from drifting away. The frequency with which this internal syntonization happens is the PTP clock correction period (PTPCLKCORP). It is a value also in the PTP clock domain, and is also rate-corrected.
To be precise, during a correction period, there is logic to determine by how many scheduler clock ticks has the PTP clock drifted. At the end of each correction period/beginning of new one, the length of a delta is shrunk or expanded with an integer number of ticks, compared with the typical 25.
So a delta lasts for 200ns (or 25 ticks) only on average. Sometimes it is longer, sometimes it is shorter. The internal syntonization logic can adjust for at most 5 ticks each 20 ticks.
The first implication is that you should choose your schedule correction period to be an integer multiple of the schedule length. Preferably one. Hence the choice in the patched linuxptp. This way, the updates are always synchronous to the transmission cycle, and therefore predictable.
The second implication is that at the beginning of a correction period, the first few deltas will be modulated in time, until the schedule engine is properly phase-aligned with the PTP clock. For this reason, you should place your best-effort traffic at the beginning of a cycle, and your time-triggered traffic afterwards.
The third implication is that once the schedule engine is started, it can only adjust for so much drift within a correction period. In the servo you can only change the PTPCLKRATE, but not step the clock (PTPCLKADD). If you want to do the latter, you need to stop and restart the schedule engine.

RikusW, 2019-03-08:
Thanks, this is quite enlightening.

vladimiroltean, 2019-03-08:
> My main concern is whether two or more cycles, each only set to update a single port, can update the gates in the same time period?

Do let me know if you come up with a way to integrate a per-port gate-control-list within the global configuration.
Not sure what software you are using, but I did implement at some point an in-kernel control of the TAS based on PTP clock: https://github.com/openil/sja1105-tool/pull/36/commits/a79990499274d3294de6f7726162ed2bc28f9320
Maybe you can use that as a starting point, and configure the GCL through something like tc-taprio and a hardware offload - that would be very interesting to see.

RikusW, 2019-03-08:
The PTP I'm using for testing is a internal rewrite of https://github.com/jdkoftinoff/gptp, I linked to libsja1105 for register access. I'll have a look at the TAS code, thanks.

vladimiroltean, 2019-03-12:
I've been exploring only the lower level aspects of PTP, so I still don't properly grasp the finer points of the protocol. But why do you need to read both PTPTSCLK and PTPCLKVAL? Can't you deduce what the uncorrected clock would have been by dividing with PTPCLKRATE, and vice-versa?

RikusW, 2019-03-12:
I thought about doing deriving the one from the other, but on second thought doing that with a servo that adjust PTPCLKRATE constantly this will not work since the point in time when PTPCLKRATE is written is unknown. The only reliable way to do this is to latch both at the same time.

RikusW, 2019-03-12:
I need both PTPTSCLK and PTPCLKVAL to calculate the offset of PTPCLKVAL from the grandmaster time, so I have the Sync receive timestamp in PTPTSCLK, having it timestamped with PTPCLKVAL as well would've been great since I also have grandmaster time when the Sync was received.
The second option is to have both latched when reading and then get the difference between Sync timestamp and the current value of PTPTSCLK, rate correct that to grandmaster clockrate and add it to the received grandmaster time and compare to PTPCLKVAL that was latched with PTPTSCLK to get the servo offset from grandmaster.

RikusW, 2019-03-12:
It seems like the PTP_CLK pin is linked to PTPTSCLK via PTPPINST ? can you confirm ?
To use PTP_CLK pin to test the servo sync it would need to be linked to PTPCLKVAL instead.

vladimiroltean, 2019-03-12:
Our PTP experts have said that the documentation is incorrect/unclear and that both PINDUR and PINST are in rate-corrected time domain. Only thing that the PTP pin does not adjust for is the phase realignment when you perform clock stepping (PTPCLKADD).

RikusW, 2019-03-12:
Thanks for clarifying. I only do PTPCLKADD once on startup before starting the scheduler.

RikusW, 2019-03-12:
The other issue I've run into was when using the "Latency and Bandwidth Tester", PTP messages don't go through anymore and then the schedulers go out of sync and everything comes apart. I think a reasonable solution would be to put PTP messages and other control traffic on priority 7. What do you think ?

vladimiroltean, 2019-03-12:
You mean the priority of multicast frames trapped to the CPU? I believe that setting is held in the HOSTPRIO field within the static config XML. Could you try?

RikusW, 2019-03-12:
I'll try that and maybe use <ip link add link eth2 name eth2.107 type vlan id 107 egress-qos-map 0:7 ingress-qos-map 7:0> and use that network interface for PTP ?

vladimiroltean, 2019-03-12:
It depends on what you're trying to do.
My above comment with HOSTPRIO is only applicable if the switch does trap the PTP frames to the CPU (I'm talking about the MAC_FLTRES and management routes mechanism). Within OpenIL this was used to implement the E2E TC, but otherwise no.
Synchronization of eth2 over PTP-unaware switch ports is not quite how things are supposed to work, and that is the reason why the synchronized Qbv demo is packaged the way it is, with PTP running over eth0. Typically you'd want to run the PTP over the swp2/swp3/swp4/swp5 SJA1105 netdevices themselves, and the Linux driver would install the multicast trapping and forwarding rules, and collect timestamps for trapped PTP frames. This way, the timestamps given to the servo would actually reflect further propagation delays such as those incurred by Qbv.
But this is not something that is packaged into OpenIL for SJA1105.
Then with Qbv you'd have to consider the effect of asymmetrical path delays, and figure out a way to only send PTP messages in-band with the Qbv window, so as to avoid these delays.
I am unclear whether you are using the patched OpenIL linuxptp for this test, or your gptp rewrite that you just linked to.

RikusW, 2019-03-12:
I'm using OpenIL with a custom gptp (802.1AS) with an interface layer to the SJA1105. The packets are sent via eth2.107 and timestamps retrieved over SPI using libsja1105, so there is only 1 cable between the switches. The LS1021A is not synced at the moment.
Getting the forwarding rules to work was a bit tricky, but fortunately there was some example code on the OpenIL Github repository. It seems like some delay on PTP packet egress is not a problem, it just takes longer to retrieve the timestamp.

closed 2019-05-06 by vladimiroltean

vladimiroltean, 2019-06-20:
FYI in the upstream PTP driver for SJA1105 i ended up implementing timestamping based on the free-running counter: https://git.kernel.org/pub/scm/linux/kernel/git/davem/net-next.git/tree/drivers/net/dsa/sja1105/sja1105_ptp.c
I plan on configuring the time-aware scheduler some time in the future using the tc-taprio offload submitted by Vinicius Costa Gomez in net-next. Actually there has been some activity on the Synopsys stmmac driver already and that can be used as an example on how to interact with this qdisc: https://lkml.org/lkml/2019/6/18/522.
Of course, right now there are no operations on the PTP hardware clock in the upstream driver yet, and those are a must if you're going to use Qbv... This is just a heads up in case you want to submit some of your work to Linux.