nxp-archive/openil_sja1105-tool #48: Invalid device id in staging area: 0xcc67e0b6

RikusW, 2019-05-14:
Hi

The sja1105-tool randomly fail to generate the configuration.

Configuration saved as /usr/lib/node_modules/lbt/./sja_config.xml.
[root@OpenIL:lbt]# ./sja_config.sh
Configuration saved as /usr/lib/node_modules/lbt/./sja_config.xml.
[root@OpenIL:lbt]# ./sja_config.sh
Invalid device id in staging area: 0xcc67e6b6
error while interpreting config
[root@OpenIL:lbt]# ./sja_config.sh
Invalid device id in staging area: 0xcc57e0b6
error while interpreting config
[root@OpenIL:lbt]# ./sja_config.sh
Invalid device id in staging area: 0xccb7deb6
error while interpreting config
[root@OpenIL:lbt]# ./sja_config.sh
Invalid device id in staging area: 0xcca7e3b6
error while interpreting config
[root@OpenIL:lbt]# ./sja_config.sh
Configuration saved as /usr/lib/node_modules/lbt/./sja_config.xml.
[root@OpenIL:lbt]# ./sja_config.sh
Configuration saved as /usr/lib/node_modules/lbt/./sja_config.xml.
[root@OpenIL:lbt]# 

[sja.zip](https://github.com/openil/sja1105-tool/files/3178259/sja.zip)

RikusW, 2019-05-14:
It does the same on the original LS1021ATSN tool and the current 80e93a7b80a7dc273b729f26caa33ef3c870ceec master branch.

RikusW, 2019-05-14:
I took out the code loading the policing json and then it works most of the time but still fails occasionally.

vladimiroltean, 2019-05-14:
I remember having seen this before, and I do have a few old paranoid patches around "guarding against concurrent access to the staging area" (the likes of flock, open with O_SYNC, fsync etc) which I also clearly remember did not fix any problem.

Trying to reproduce it now, I left this running for a few hours and I couldn't:

```
[root@OpenIL:~]# while :; do ./sja_config.sh; done
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
Configuration saved as ./sja_config.xml.
View with: "sja1105-tool config load ./sja_config.xml; sja1105-tool config show | less"
(...)
```

Unfortunately I really can't remember what was the problem in my case, but these things come to mind:

Are your libsja1105.so and the sja1105-tool versions in sync?
Can you prevent the NETCONF server from running, if you haven't already (/etc/init.d/S90netconf stop; rm /etc/init.d/S90netconf)?

Otherwise we may try to debug.

The device id makes up the first 4 bytes in the /lib/firmware/sja1105.bin file, so it appears to get corrupted very early on during the write.
Because the bash script runs with "-e", it exists as soon as the first error is encountered. That means that "hexdump /lib/firmware/sja1105.bin" should hopefully yield the full corrupted binary. Could you attach that to see what it contains? Or maybe the file is valid when hexdumping it, case in which it may be a read problem.
One other thing is to add the "-x" option to the script, so as to see exactly which script line fails (is it always the same command or is it random?).

RikusW, 2019-05-16:
The xml seems to load fine but I suspect config modify, this is also very slow on the LS1021 when a 44 modifications are done on the ingress policy table, try running the sja_config.sh instead.

vladimiroltean, 2019-05-16:
Isn't that what I ran? `while :; do ./sja_config.sh; done`

RikusW, 2019-05-16:
Sorry, overlooked that line...

RikusW, 2019-05-16:
++ jq .smax
+ sja1105-tool config modify 'l2-policing-table[9]' smax 65535
+ tool 9 rate '{
  "port": 1,
  "priority": "1",
  "sharindx": 9,
  "smax": 65535,
  "rate": 64000,
  "maxlen": 1522,
  "partition": 0
}'
++ jq .rate
+ sja1105-tool config modify 'l2-policing-table[9]' rate 64000
+ tool 9 maxlen '{
  "port": 1,
  "priority": "1",
  "sharindx": 9,
  "smax": 65535,
  "rate": 64000,
  "maxlen": 1522,
  "partition": 0
}'
++ jq .maxlen
+ sja1105-tool config modify 'l2-policing-table[9]' maxlen 1522
Invalid device id in staging area: 0xcce7dbb6
error while interpreting config
[sjabin.zip](https://github.com/openil/sja1105-tool/files/3186062/sjabin.zip)

RikusW, 2019-05-16:
0000000 009e 0e03 0006 0000 0000 5a00 d24e 0fc5
0000010 f702 0000 ff03 e8ff f702 0000 ff07 e8ff
0000020 f902 0000 ff0b e8ff f902 0000 ff0f e8ff
0000030 f902 0000 ff13 e8ff f902 0000 ff17 e8ff
0000040 f902 0000 ff1b e8ff f702 0000 ff1f e8ff
0000050 f702 0000 ff23 e8ff f702 0000 ff27 e8ff
0000060 f702 0000 ff2b e8ff f702 0000 ff2f e8ff
0000070 f702 0000 ff33 e8ff f702 0000 ff37 e8ff
0000080 f702 0000 ff3b e8ff f702 0000 ff3f e8ff
0000090 f702 0000 ff43 e8ff f702 0000 ff47 e8ff
00000a0 f702 0000 ff4b e8ff f702 0000 ff4f e8ff
00000b0 f702 0000 ff53 e8ff f702 0000 ff57 e8ff
00000c0 f702 0000 ff5b e8ff f702 0000 ff5f e8ff
00000d0 f702 0000 ff63 e8ff f702 0000 ff67 e8ff
00000e0 f702 0000 ff6b e8ff f702 0000 ff6f e8ff
00000f0 f702 0000 ff73 e8ff f702 0000 ff77 e8ff
0000100 f702 0000 ff7b e8ff f702 0000 ff7f e8ff
0000110 f702 0000 ff83 e8ff f702 0000 ff87 e8ff
0000120 f702 0000 ff8b e8ff f702 0000 ff8f e8ff
0000130 f702 0000 ff93 e8ff f702 0000 ff97 e8ff
0000140 f702 0000 ff9b e8ff f702 0000 ff9f e8ff
0000150 0000 0000 0000 0000 0000 0000 0000 0000
*
0000170 0000 0000 0000 0000 3603 ad47 0007 0000
0000180 0000 0200 0b7d f2cb 0000 0000 3f00 00f0
0000190 3888 8586 0008 0000 0000 1a00 f66a 5323
00001a0 0010 0000 bdf7 8df5 0010 0000 7bef 8df5
00001b0 0010 0000 f7de 8df5 0010 0000 efbd 8df5
00001c0 0010 0000 df7b 8df5 0000 0000 0000 0000
00001d0 0092 0000 0000 2400 0024 0000 0000 4900
00001e0 00b6 0000 0000 6d00 0048 0000 0000 9200
00001f0 00da 0000 0000 b600 006c 0000 0000 db00
0000200 00fe 0000 0000 ff00 4267 06e0 0009 0000
0000210 0000 2300 b5da c8bd 00fe 0e00 0000 0100
0000220 fc08 0201 40bf e44f 17e0 02fb ff06 cf40
0000230 f0ff fc3b 00fe 0e00 0000 0100 fc08 0201
0000240 40bf e44f 17e0 02fb ff06 cf40 f0ff fc3b
0000250 00fe 0e00 0000 0100 fc08 0201 40bf e44f
0000260 17e0 02fb ff06 cf40 f0ff fc3b 00fe 0e00
0000270 0000 0100 fc08 0201 40bf e44f 17e0 02fb
0000280 ff06 cf40 f0ff fc3b 00fe 0e00 0000 0100
0000290 fc08 0201 40bf e44f 17e0 02fb ff06 cf40
00002a0 f0ff fc3b e2fc a35e 000d 0000 0000 0100
00002b0 0e25 bd7c 0100 e025 a6d0 6e2b 000e 0000
00002c0 0000 0300 a7c8 e6ce 7400 0020 0000 0000
00002d0 0000 0000 eb02 5a6c 0011 0000 0000 0a00
00002e0 1f57 3f81 4406 0000 0000 0804 0000 0000
00002f0 0dff 00a0 ffff ffff ffff ffff ff00 ffff
0000300 0000 0000 0000 0000 0058 0000 e804 759c
0000310 004e 0000 0000 0100 5d3a 245e 6c5b 0000
0000320 b514 cb73 0000 0000 0000 0000 adde efbe
0000330

RikusW, 2019-05-16:
I stopped netconf and now it works, thanks. It still takes about 7 seconds, any idea why its so slow ?

vladimiroltean, 2019-05-16:
I haven't really looked at it with perf or anything, but the tool is not particularly designed with efficiency in mind. And also it's basically in maintenance mode now, I suppose OpenIL will switch at some point to use the mainline driver once it gets support for PTP timestamping and TTEthernet via tc-taprio.

Good to know that stopping NETCONF did the trick. The netopeer-server has a inotify listener for changes to the staging area file and it is also attempting to write to it when it detects changes. I believe it's written somewhere in the documentation that auto-flushing should be enabled in sja1105.conf if both are to be used at the same time.

closed 2019-05-16 by RikusW