nxp-archive/openil_sja1105-tool #49: [Bug] Broken link for downloding the gcc component from linaro cross-toolchain

JayachandranRameshBabu, 2019-07-23:
Hello,
It seems like the web link to download the packages(gcc) from linaro cross-toolchain has been updated. The attached patch here fixes the README.md 

[ReadMe_Patch.zip](https://github.com/openil/sja1105-tool/files/3422310/ReadMe_Patch.zip)

Regards,
Jayachandran

vladimiroltean, 2019-07-23:
Hi Jayachandran,

Please submit a normal pull request.
Also please take the opportunity to point to the latest toolchain binaries which currently means gcc 7.4.1: https://releases.linaro.org/components/toolchain/binaries/latest-7/arm-linux-gnueabihf/

Regards,
-Vladimir

JayachandranRameshBabu, 2019-07-24:
Hello Vladimir,

Thanks for your comments. Please find the pull request #50 

Regards,
Jayachandran Ramesh Babu

closed 2019-07-25 by vladimiroltean