nxp-archive/openil_sja1105-tool #54: Synchronise TSN schedule with other devices

LehrChristoph, 2019-09-16:
Hi,
I'm currently working on synchronising the SJA1105 with another TSN-device. Since the SJA1105 does not support ADMIN configuration, I'm curious of the following statement:
"The schedule starts when the local clock of the synchronization algorithm
wraps with the entry point as indexed by the integration cycle number contained in the
PCF used for integration. "
As far as I understood that statement, the schedule starts when the local clock reaches DELTA, but when is the wrap around and how is this synchronised with PTP?

The reason for asking this, if I understand this I may be possible to better syncrhonise the schedules of my devices.

Many Thanks for your help.
BR Christoph

vladimiroltean, 2019-09-16:
Hi Cristoph,

Have you looked at the [community kernel](https://github.com/vladimiroltean/ls1021atsn-linux/commits/master)?
The drivers/net/dsa/sja1105/sja1105_tas.c file has a detailed explanation about how the hardware scheduler works. It can be configured from user space through a tc-taprio offload. The gate events have a resolution of 200 ns in PTP time and you need to take care that no 2 gate events trigger at the same time on multiple ports.

Hope this helps,
-Vladimir

LehrChristoph, 2019-09-18:
I use PTP to synchronise the whole network, so I'd need something compareable to the admin Basetime, since Cycle time should be the same.

I checked the code, it helped me a little bit, but what does PTPSCHTM stand for?

Is there some API or event I'd could subscribe to, to synchronise my application to the network schedule?

BR Christoph

vladimiroltean, 2019-09-18:
Hi Cristoph,

I've been using a script that connects through SSH to all devices in the network and installs the schedule through tc-taprio using the same base-time. I hope I will be able to publish a more ready-made example in the next days/weeks. Anyway you can see the [driver documentation for tc-taprio](https://github.com/vladimiroltean/ls1021atsn-linux/blob/master/Documentation/networking/dsa/sja1105.rst#time-aware-scheduling) for a basic example.

At the moment I know of no API, although I would be very interested to find out too. Fundamentally it depends a lot on whether you're looking to see something dynamic/plug-and-play (like the MSRP from AVB) or something centralized (like NETCONF for industrial networking). OpenIL has some NETCONF examples, but none uses tc-taprio as a backend. And Open-AVB has some examples as well, but neither does that. Maybe you can explain a bit more about what you want to do.

You can also follow [this discussion](https://github.com/openil/openil/issues/23) since there are other people as well who are just setting up the PTP-enabled kernel, and there are already a few pointers in there. We can keep this open for discussion related to network schedule management.

Hope this helps,
-Vladimir