nxp-archive/openil_sja1105-tool #55: Schedule configuration problem with sja1105-tool

jihe123, 2019-11-07:
Hello, i am using the LS1021ATSN-PA board(just one),and i did one board demo according to the PDF(Open Industrial Linux User Guide Release v0.2),but when i did the schedule configuration(6.8.6), there's something wrong like this:
[root@OpenIL:~]# sja1105-tool conf mod schedule-table entry-count 2
[root@OpenIL:sja1105]# for i in 0 1; do sja1105-tool conf mod schedule-table[$i]
 \destports 0b00100;done
Index out of bounds!
Please adjust the entry count of the table:
* config modify <table> entry-count <value>)
modify failed!
Index out of bounds!
Please adjust the entry count of the table:
* config modify <table> entry-count <value>)
modify failed!
Do you know what's wrong with this? Thank you .

vladimiroltean, 2019-11-07:
Why are you using OpenIL v0.2 when 1.6 is available? Is the sja1105-tool version also from v0.2?

jihe123, 2019-11-07:
Because there's only one ls1021atsn board and the one board demo introduction from v1.6  is so simple,and i cannot understand where is or how to creat the ' ~/.ssh/authorized_hosts configuration ' file ,because there is a space between authorized_hosts and configuration,so i found v0.2 and did the demo.
There is a question, i want to know ,could i control the sja1105 without the sja1105-tool in the openil,just like fpga.

vladimiroltean, 2019-11-07:
Hold on, let's take the items one by one.
The file is called `authorized_hosts` and is located in the `.ssh` folder of the current user's home, which is also abbreviated as `~` in Linux. So the full path is `~/.ssh/authorized_hosts`. There is plenty of information on the web on how to configure a SSH server and client and I don't intend to go into further detail on that here.

> could i control the sja1105 without the sja1105-tool in the openil,just like fpga.

What do you mean "just like fpga", how do you control a FPGA?
If I ignore the latter part of the question and concentrate on the first part ("without sja1105-tool"), yes, the switch is supported in the mainline Linux kernel and the abstractions used there are completely different from what sja1105-tool offers (DSA for switch management, switch ports as individual net devices, tc-taprio for Qbv configuration, etc). A high-level documentation for the mainline Linux solution is here: https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html
Is this what you mean by your question?

jihe123, 2019-11-08:
Thank you for your answer.
I have known the reason about the problem "[root@OpenIL:~]# sja1105-tool conf mod schedule-table entry-count 2
[root@OpenIL:sja1105]# for i in 0 1; do sja1105-tool conf mod schedule-table[$i]" , it's because the commond " [root@OpenIL:~]# sja1105-tool conf mod schedule-table entry-count 2" did not work,the schedule-table is empty, but i do not know why.
There are still questions about the LS1021ATSN-PA board:
Can  ls1021a(ARM), CPLD and SJA1105 work together?
Is it possible to program the LS1021A with the IDE(Codewassior), then control the SJA1105.
Thany you again.

vladimiroltean, 2019-11-08:
There are examples available [[1]](https://github.com/openil/sja1105-tool/blob/master/src/helpers/configs/synchronized-qbv/single-qbv-standalone-clk.sh) [[2]](https://github.com/openil/sja1105-tool/blob/master/src/helpers/bin/scheduler-create), have you tried running them first and see what is different in your command? I wouldn't recommend using sja1105-tool from version 0.2, please try the newest versions. There have been multiple bug fixes in the past years.

> Can ls1021a(ARM), CPLD and SJA1105 work together?

The CPLD is not intended to be programmable. It only serves to drive the reset logic of the board, multiplex interrupt lines, etc. Furthermore, the CPLD on the LS1021A should be accessed through the IFC bus, but due to pinmuxing limitations in the LS1021A RCW design, IFC is mutually exclusive with QSPI. The RCW binaries distributed with OpenIL have preferred to choose QSPI access, so the IFC bus (and therefore the CPLD) is not accessible currently. And to be fair, I don't see any major reason to access it.

> Is it possible to program the LS1021A with the IDE(Codewassior),

CodeWarrior for ARMv7 supports the LS1021A SoC but not the TSN board in particular. I don't think it offers any particular benefit.

> then control the SJA1105.

Again, what do you mean by "control"? The SJA1105 is a discrete SPI device, and CodeWarrior only has insight into the DSPI controller registers, at most. It does not have any sort of support for SJA1105.

jihe123, 2019-11-08:
Thank you , i will try the examples .
 I know the v0.2 is too old, but i can not find another one like v0.2 , has a detailed introduction. 
Thank you again.

vladimiroltean, 2019-11-08:
The documentation in later releases has become less verbose because more of the examples have been moved from the PDF into the scripts I linked to.

vladimiroltean, 2019-11-08:
For your information, you will find some limitations in the sja1105-tool, like the fact that PTP over the switch ports is not supported. You may still use the OpenIL documentation for a description of the hardware capabilities, but if you want PTP you can try the ["community"](https://github.com/vladimiroltean/openil-community) (unofficial) version of OpenIL that supports that.

jihe123, 2019-11-08:
Thank you, for the examples of single-qbv-standalone-clk.sh and scheduler-create,the first one generate a .xml file, what can i do with it? (load it to the board? )
The second, scheduler-create, it just stop there and generate nothing.
"j123dadi@123dadi:~/sja1105-tool/src/helpers/bin$ scheduler-create

"
Is there anything wrong? 
Thank you again for your patience and kind.

vladimiroltean, 2019-11-08:
For the first example, you transfer the xml to the board (or run the commands directly on the board) and run "sja1105-tool config load -f file.xml".
The second example is the actual script that is being run to generate the xml. You are not supposed to run it directly. You will find it called from the first script.

jihe123, 2019-11-08:
Ok, i will try it later, thank you for your answer.

jihe123, 2019-11-08:
Have a great day☀️

liu-jeff, 2019-11-20:
> 祝你有美好的一天☀️

> Have a great day☀️

Hello, do you have a demo tutorial for a single LS1021 development board?

jihe123, 2019-11-20:
> > 祝你有美好的一天sunny
> 
> > Have a great daysunny
> 
> Hello, do you have a demo tutorial for a single LS1021 development board?

liu-jeff, 2019-11-20:
> > > 祝你有美好的一天sunny
> > 
> > 
> > > Have a great daysunny
> > 
> > 
> > Hello, do you have a demo tutorial for a single LS1021 development board?

?

jihe123, 2019-11-20:
> > > > 祝你有美好的一天sunny
> > > 
> > > 
> > > > Have a great daysunny
> > > 
> > > 
> > > Hello, do you have a demo tutorial for a single LS1021 development board?
> 
> ?

回复错了，我用的是ls1021atsn那块板子，然后找了一个以前版本的文档，里面有一块板子的demo指导。

liu-jeff, 2019-11-20:
> > > > > 祝你有美好的一天sunny
> > > > 
> > > > 
> > > > > Have a great daysunny
> > > > 
> > > > 
> > > > Hello, do you have a demo tutorial for a single LS1021 development board?
> > 
> > 
> > ?
> 
> 回复错了，我用的是ls1021atsn那块板子，然后找了一个以前版本的文档，里面有一块板子的demo指导。
你会中文啊？

jihe123, 2019-11-20:
> > > > > > 祝你有美好的一天sunny
> > > > > 
> > > > > 
> > > > > > Have a great daysunny
> > > > > 
> > > > > 
> > > > > Hello, do you have a demo tutorial for a single LS1021 development board?
> > > 
> > > 
> > > ?
> > 
> > 
> > 回复错了，我用的是ls1021atsn那块板子，然后找了一个以前版本的文档，里面有一块板子的demo指导。
> > 你会中文啊？

中国人不会中文啊。。。

jihe123, 2019-11-20:
> 
> 
> > > > > > 祝你有美好的一天sunny
> > > > > 
[NXP Digital Networking Global Software Development Open Industrial Linux User Guide Release v0.2.pdf](https://github.com/openil/sja1105-tool/files/3868280/NXP.Digital.Networking.Global.Software.Development.Open.Industrial.Linux.User.Guide.Release.v0.2.pdf)

> > > > > 
> > > > > > Have a great daysunny
> > > > > 
> > > > > 
> > > > > Hello, do you have a demo tutorial for a single LS1021 development board?
> > > 
> > > 
> > > ?
> > 
> > 
> > 回复错了，我用的是ls1021atsn那块板子，然后找了一个以前版本的文档，里面有一块板子的demo指导。
> > 你会中文啊？
我上传了，你看一下，位置有点不对，在上面。

liu-jeff, 2019-11-20:
> [NXP Digital Networking Global Software Development Open Industrial Linux User Guide Release [...]

非常感谢，，加个qq交流一波呗 1579530627

closed 2019-11-28 by jihe123