nxp-archive/openil_sja1105-tool #56: How to configure LS1021ATSN with sja1105-tool , so that vlan tag of export stream is not deleted

liu-jeff, 2019-11-22:
How to configure LS1021ATSN with sja1105-tool , so that vlan tag of export stream is not deleted

vladimiroltean, 2019-11-22:
You need to edit the `vlan-lookup-table` from your sja1105-tool xml and set the corresponding bits of `tag_port` for your ports on which you want the VLAN to be sent as tagged.
Man page is [here](https://github.com/openil/sja1105-tool/blob/master/docs/md/sja1105-tool-config-format.5.md#vlan-lookup-table).
Switch manual is [here](https://www.nxp.com/docs/en/user-guide/UM10944.pdf).

liu-jeff, 2019-11-22:
> You need to edit the `vlan-lookup-table` from your sja1105-tool xml and set the corresponding bits of `tag_port` for your ports on which you want the VLAN to be sent as tagged.
> Man page is [...]
first,very thank you,Can you give me an example? I still haven't figured out how to use tag_port to make my ETH3 exit stream a stream with a valn tag.

vladimiroltean, 2019-11-22:
First of all, the ETH3 label on the chassis is just a label. The switch port numbering is completely different, see [here](https://github.com/openil/sja1105-tool#port-numbering-on-the-nxp-ls1021atsn-board). You need to use the number of the chip port.
The default sja1105-tool xml puts a pvid of 0 for each port, and only adds VLAN 0 to the VLAN table. You can start by making tag_port equal to 0x1f, aka make all ports send traffic as tagged, to make sure the feature works. Then you can only enable BIT(2) (aka 0x4), for your ETH3 port.

liu-jeff, 2019-11-22:
> First of all, the ETH3 label on the chassis is just a label. The switch port numbering is completely different, see [here](https://github.com/openil/sja1105-tool#port-numbering-on-the-nxp-ls1021atsn-board). You need to use the number of the chip port.
> The default sja1105-tool xml puts a pvid of 0 for each port, and only adds VLAN 0 to the VLAN table. You can start by making tag_port equal to 0x1f, aka make all ports send traffic as tagged, to make sure the feature works. Then you can only enable BIT(2) (aka 0x4), for your ETH3 port.
> Then you can add more VLANs to the VLAN table.

I can now grab the vlan package, but it seems that iperf, ping these will not work, is it necessary to remove the vlan tag for the three or more packages?

vladimiroltean, 2019-11-22:
You need to add a VLAN sub-interface with "ip link" on eth2 to decode the VLAN ID, if that is different than 0. If it is zero you should not need to do that.
For iperf, you also need to increase the MTU of the switch ports (MAXLEN field of the 45 policer entries) with 4 (the length of the VLAN tags).

liu-jeff, 2019-11-22:
> You need to add a VLAN sub-interface with "ip link" on eth2 to decode the VLAN ID, if that is different than 0. If it is zero you should not need to do that.
> For iperf, you also need to increase the [...]

So how do you set the MTU of the port?

vladimiroltean, 2019-11-22:
Like I just said, search for MAXLEN in the xml and increase it by 4.

liu-jeff, 2019-11-22:
[image]
you look ,next what can i do?

vladimiroltean, 2019-11-22:
sja1105-tool config save tmp.xml
Replace 0x5ee (1518 bytes) with 0x5f2 (1522 bytes) in tmp.xml
sja1105-tool config load -f tmp.xml

liu-jeff, 2019-11-22:
> sja1105-tool config save tmp.xml
> Replace 0x5ee (1518 bytes) with 0x5f2 (1522 bytes) in tmp.xml
> sja1105-tool config load -f tmp.xml
Can you modify it by command?

vladimiroltean, 2019-11-22:
```
for port in $(seq 0 4); do
    for prio in $(seq 0 7); do
        sja1105-tool config modify l2-policing-table[$((${port}*8 + ${prio}))] maxlen 1522
    done
done
sja1105-tool config upload
```

liu-jeff, 2019-11-22:
> ```
> for port in $(seq 0 4); do
>     for prio in $(seq 0 7); do
>         sja1105-tool config modify l2-policing-table[$((${port}*8 + ${prio}))] maxlen 1522
>     done
> done
> sja1105-tool config upload
> ```

I do it ,but Iperf and ping still can't work

vladimiroltean, 2019-11-22:
Please share port counters.

liu-jeff, 2019-11-22:
> Please share port counters.

[image]
[image]
[image]

Now, my stream outlet is ETH3, so I set the tag_port to 2, but I didn't catch the vlan package on the ETH3 exit computer.

vladimiroltean, 2019-11-22:
Those are not port counters.
`sja1105-tool status port`.

liu-jeff, 2019-11-22:
> sja1105-tool status port

[image]

liu-jeff, 2019-11-22:
> Those are not port counters.
> `sja1105-tool status port`.
[image]
Now, I don’t even have the vlan bag.

vladimiroltean, 2019-11-22:
What is connected to port 1 (ETH2)? Your sender machine, presumably? It says N_CRCERR 706. What link speed is it running at? Gigabit or 100 mbps? Have you seen the "Link speed autonegotiation is not supported." [known issue](https://github.com/openil/sja1105-tool#known-issues)?

liu-jeff, 2019-11-22:
> What is connected to port 1 (ETH2)? Your sender machine, presumably? It says N_CRCERR 706. What link speed is it running at? Gigabit or 100 mbps? Have you seen the "Link speed autonegotiation is not supported." [known issue](https://github.com/openil/sja1105-tool#known-issues)?

port1(ETH2) is my sender,It is 10000mbps，now,it do not have vlan tag.
[image]
I didn't do this step, I don't know how to do it.

vladimiroltean, 2019-11-22:
Once you get rid of the CRC errors:
```
sja1105-tool config modify vlan-lookup-table[0] $((1 << 2))
```

liu-jeff, 2019-11-22:
> Once you get rid of the CRC errors:
> 
> ```
> sja1105-tool config modify vlan-lookup-table[0] $((1 << 2))
> ```

[image]
it error

vladimiroltean, 2019-11-22:
Please check the updated response....

liu-jeff, 2019-11-22:
> Please check the updated response....
after command ,it do not have vlan tag too

vladimiroltean, 2019-11-22:
How do you check?

liu-jeff, 2019-11-22:
> Please check the updated response....

Is it because I have no settings?
[image]

liu-jeff, 2019-11-22:
> How do you check?

I do not kown,I don't know, but I didn't do this because I don't know how to do this step.how to enable BIT(2)(aks 0x4),for ETH3 port

vladimiroltean, 2019-11-22:
You just did that.
BIT(2) means 1 << 2. Label ETH3 is port 2.

I meant to ask how do you know the frames are not VLAN-tagged.

liu-jeff, 2019-11-22:
> You just did that.
> BIT(2) means 1 << 2. Label ETH3 is port 2.
> 
> I meant to ask how do you know the frames are not VLAN-tagged.

I picked up a host on ETH3, as an iperf server, and then grabbed the packet through wireshark (already set to grab the vlan package), but did not catch the vlan tag package.

vladimiroltean, 2019-11-22:
If you set tag_port to 0x1f do you see the VLAN tag?

liu-jeff, 2019-11-22:
> If you set tag_port to 0x1f do you see the VLAN tag?

after it , Iperf can't

liu-jeff, 2019-11-22:
> > If you set tag_port to 0x1f do you see the VLAN tag?
> 
> after it , Iperf can't

> If you set tag_port to 0x1f do you see the VLAN tag?

[image]
[image]

liu-jeff, 2019-11-22:
> If you set tag_port to 0x1f do you see the VLAN tag?

ETH2(1) is my iperf sender ,ETH3(2) is my iperf server,
[image]
[image]
[image]
[image]
From the screenshot above, I feel that I have not made a mistake. What do you think is wrong?

vladimiroltean, 2019-11-22:
I'm sorry, isn't there a VLAN tag with an ID of 0 and a PCP of 7 in your wireshark dump?

liu-jeff, 2019-11-22:
> I'm sorry, isn't there a VLAN tag with an ID of 0 and a PCP of 7 in your wireshark dump?

I know ，  You can take a look at the screenshot above. The package I sent from iperf is vlan proi 3 not 7,I think there are lots of problems

vladimiroltean, 2019-11-22:
Are you sending VLAN-tagged traffic or untagged?
If it is untagged, it gets tagged by the switch on the ingress port, with the MAC's VLAN ID and VLAN PRIO.
If it is tagged, the tag from the original frame is kept.

liu-jeff, 2019-11-23:
> Are you sending VLAN-tagged traffic or untagged?
> If it is untagged, it gets tagged by the switch on the ingress port, with the MAC's VLAN ID and VLAN PRIO.
> If it is tagged, the tag from the original frame is kept.

I am sending untagged,   i know on the ingress port ,it is combined with vlan prio, on egress,it will be deleted,but i want to reserve the tag on egress , According to the above configuration, it seems that this effect can not be achieved

liu-jeff, 2019-11-24:
> Are you sending VLAN-tagged traffic or untagged?
> If it is untagged, it gets tagged by the switch on the ingress port, with the MAC's VLAN ID and VLAN PRIO.
> If it is tagged, the tag from the original frame is kept.
Can you give me some suggestions

vladimiroltean, 2019-11-24:
> Can you give me some suggestions

I can't follow your issue, sorry.
The switch tags all untagged frames with an internal VLAN (mac-config-table[vlanid]). By default in sja1105-tool, that VLAN ID is 0.
The VLAN ID of 0 is removed on egress, via vlan-table[tag_port]. So traffic that was untagged at input is still untagged at output.
What you are doing when you are making VID 0 as egress-tagged is that you are making the port-based VLAN (pvid) untagged input and tagged output. So you break untagged communication. In the wireshark screenshot you sent, the board sees the ARP request but the other device doesn't see the ARP reply, because it's not expecting a VLAN tag.
Linux hosts can actually treat 802.1p (VID 0) tagged traffic as untagged, if they have the 8021q module inserted, which adds VID 0 automatically in their rx filter. This probably did not happen in your case though.
So what you need to do, if you want tagged traffic on output, is pick a valid port-based VLAN for the MACs. Arguably the sja1105-tool default of 0 is not very good. You should add the VLAN ID of 1 to your xml and change all ports to tag by default with 1. Then VLAN 0 becomes available for transmitting tagged frames only.
Then you need to think about how your operating system on the sender and the receiver is going to process that VLAN. That is completely outside the scope of this issue, but it might cause you problems nonetheless.
Then, once all of that is figured out and the switch is configured, you look at changes in the port statistics counters to figure out where and why the switch drops frames. It may or may not be related to VLANs.
Once that is clear as well, you should formulate more clearly what your issue is. I think you achieved the stated goal of not removing the VLAN tag for VID 0 on egress, and that we're talking about something else here.

liu-jeff, 2019-11-24:
> I can't follow your issue, sorry.
> The switch tags all untagged frames with an internal VLAN (mac-config-table[vlanid]). By default in sja1105-tool, that VLAN ID is 0.
> The VLAN ID of 0 is removed on [...]
So I need to change the vlan id for the stream of the ingress? Like 3 not 0

vladimiroltean, 2019-11-24:
> So I need to change the vlan id for the stream of the ingress? Like 3 not 0

I only said it needs to be different than the port-based default. You can change either of them.

liu-jeff, 2019-11-24:
> > So I need to change the vlan id for the stream of the ingress? Like 3 not 0
> 
> I only said it needs to be different than the port-based default. You can change either of them., I will try it.

liu-jeff, 2019-11-24:
> > So I need to change the vlan id for the stream of the ingress? Like 3 not 0
> 
> I only said it needs to be different than the port-based default. You can change either of them.

Thank you ,I will try it!

closed 2019-11-24 by liu-jeff

liu-jeff, 2019-11-24:
> > So I need to change the vlan id for the stream of the ingress? Like 3 not 0
> 
> I only said it needs to be different than the port-based default. You can change either of them.
I changed a vlan id at the entrance and found that it was directly unreasonable.