nxp-archive/openil_sja1105-tool #57: How to set vlan id at the ingress

liu-jeff, 2019-11-24:
After setting the vlan id at the ingress port, the host connected to the ingress cannot communicate with the board directly.

vladimiroltean, 2019-11-24:
Have you added the new VID to the vlan-lookup-table? Just create another entry based on the one for VID 0, and replace vlanid with your number.

liu-jeff, 2019-11-24:
> Have you added the new VID to the vlan-lookup-table? Just create another entry based on the one for VID 0, and replace vlanid with your number.
could you please give me the command to create another entry ,and how to modify vlanid and tag_port in the case of multiple entries

vladimiroltean, 2019-11-24:
The easiest way is to export the config to a temporary xml, copy and paste, and load the new xml.

liu-jeff, 2019-11-24:
> The easiest way is to export the config to a temporary xml, copy and paste, and load the new xml.

Could be more detailed?

liu-jeff, 2019-11-24:
> The easiest way is to export the config to a temporary xml, copy and paste, and load the new xml.

how to export the config to a temporary xml ,

vladimiroltean, 2019-11-24:
Absolutely the same technique as my previous answer https://github.com/openil/sja1105-tool/issues/56#issuecomment-557522990.
Edit the tmp.xml with a text editor and add a new entry to vlan-lookup-table.

liu-jeff, 2019-11-24:
> Absolutely the same technique as my previous answer [#56 (comment)](https://github.com/openil/sja1105-tool/issues/56#issuecomment-557522990).
> Edit the tmp.xml with a text editor and add a new entry to vlan-lookup-table.

thank you very much ,but egress also has no valn_tag
[image]

vladimiroltean, 2019-11-24:
What type of traffic are you sending to the switch, and what do you expect it to do with it?
My understanding is that you have changed the port-based default VLAN to 5, but you are still sending untagged traffic.
You should have left tag_port as 0 for the port-based default VLAN, and 0x1f for the egress tagged VLAN, not the other way around.

liu-jeff, 2019-11-24:
> What type of traffic are you sending to the switch, and what do you expect it to do with it?
> My understanding is that you have changed the port-based default VLAN to 5, but you are still sending [...]

[image]
[image]
you see, port 1 is my sender, and was set vlanid 5, I am sending untagged traffic,but in ingress, it will be tagged vlanid 5 ,but it still nowhere

vladimiroltean, 2019-11-24:
Be more clear. What is "still nowhere", you don't get the frame at all, or you get it, but untagged? Which switch port counters increment?

liu-jeff, 2019-11-24:
> Be more clear. What is "still nowhere", you don't get the frame at all, or you get it, but untagged? Which switch port counters increment?

ping the board ,that say Destination host Unreachable

vladimiroltean, 2019-11-24:
You send an untagged packet on switch port 1.
It gets tagged with the port-based default of 5.
It is flooded on all ports, including port 4 which is the CPU port.
The tag_port bit 4 of VLAN 5, for the CPU port, is set, so the frame is transmitted to the CPU as tagged.
There is a VLAN sub-interface of eth2 with an IP address configured. It receives the ARP packet, recognizes its IP, and responds.
The response goes back through the switch, already tagged with VID 5 by the VLAN sub-interface of eth2.
The switch floods it on all ports, including port 1.
The bit 1 of tag_port is set, so the VLAN tag is kept while transmitting the response to your device.
The device expects untagged traffic and doesn't know what to do with the VLAN tag.
How can this possibly work?
Do you even read my replies?

liu-jeff, 2019-11-24:
> Be more clear. What is "still nowhere", you don't get the frame at all, or you get it, but untagged? Which switch port counters increment?

I am sending untagged ping message form host1(connected with port 1) to host3(the board) and host2(connected with port 2)，when I only modify the vlanpro of port 1 ,it can communicate normally,but when I modify the vlan id of the port1 to 5, it says Destination host Unreachable

vladimiroltean, 2019-11-24:
Well, then don't.
Try to understand what you are doing. You need an untagged VLAN that is untagged on all ports, and your tagged traffic to use another VLAN.

liu-jeff, 2019-11-24:
> Well, then don't.
> Try to understand what you are doing. You need an untagged VLAN that is untagged on all ports, and your tagged traffic to use another VLAN.

The bit 1 of tag_port is set, so the VLAN tag is kept while transmitting the response to your device.
The device expects untagged traffic and doesn't know what to do with the VLAN tag.
How can this possibly work?
Do you even read my replies?

> Well, then don't.
> Try to understand what you are doing. You need an untagged VLAN that is untagged on all ports, and your tagged traffic to use another VLAN.

I am so sorry to bother you and very thank you . please let me think it over.

liu-jeff, 2019-11-24:
> Well, then don't.
> Try to understand what you are doing. You need an untagged VLAN that is untagged on all ports, and your tagged traffic to use another VLAN.
I seem to understand that the host cannot handle vlan packets (without 802.1q module). Thank you very much. I will try again tomorrow.

closed 2019-11-25 by liu-jeff