nxp-archive/openil_sja1105-tool #58: Does sja1105-tool send the configuration directly to the sja1105 chip?

liu-jeff, 2019-11-28:
Hello, I want to know that sja1105-tool sends configuration through Ioctl function. Is there a receiving function in kernel mode?

vladimiroltean, 2019-11-28:
sja1105-tool uses the generic SPI driver (spidev), which is both its best feature and its worst weakness.
There also exists a native kernel driver which makes the switch look much more similar to Felix from LS1028A (net devices for switch ports, PHY management, VLAN and FDB manipulation with standard iproute2 tools, etc etc).
Why ask?

liu-jeff, 2019-11-28:
> Hello, I want to know that sja1105-tool sends configuration through Ioctl function. Is there a receiving function in kernel mode?

> sja1105-tool uses the generic SPI driver (spidev), which is both its best feature and its worst weakness.
> There also exists a native kernel driver which makes the switch look much more similar to [...]
I was reading the code of openil, and only found the interface of Netlink in the kernel, but not the interface of ioctl.

vladimiroltean, 2019-11-28:
This is the kernel driver used by sja1105-tool:
https://github.com/openil/linux/blob/master/drivers/spi/spidev.c

liu-jeff, 2019-11-28:
> This is the kernel driver used by sja1105-tool:
> https://github.com/openil/linux/blob/master/drivers/spi/spidev.c

Thank you very much, so is the qbv function in LS1021ATSN implemented in the kernel or sja1105?

vladimiroltean, 2019-11-28:
It is implemented in hardware.
The configuration is done from sja1105-tool.
In the native kernel driver I was mentioning, it is done through the standard tc-taprio offload command (still not tsntool).

liu-jeff, 2019-11-28:
> In the native kernel driver I was mentioning, it is done through the standard tc-taprio offload command (still not tsntool).
So for LS1021ATSN, if it forwards the data packet, will it not go to the kernel? Is this packet not extracted in the kernel?

liu-jeff, 2019-11-28:
> It is implemented in hardware.
> The configuration is done from sja1105-tool.
> In the native kernel driver I was mentioning, it is done through the standard tc-taprio offload command (still not tsntool).

In fact, I just want to extract the data packets forwarded by the switch in the kernel and perform some operations on them.

liu-jeff, 2019-11-28:
> It is implemented in hardware.
> The configuration is done from sja1105-tool.
> In the native kernel driver I was mentioning, it is done through the standard tc-taprio offload command (still not tsntool).
the standard tc-taprio offload command can achieve qbv?

vladimiroltean, 2019-11-28:
Yes, see [the documentation](https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html#time-aware-scheduling) for it.

liu-jeff, 2019-11-28:
> Yes, see [the documentation](https://www.kernel.org/doc/html/latest/networking/dsa/sja1105.html#time-aware-scheduling) for it.

But can your native kernel code be ported to openil?