nxp-archive/openil_sja1105-tool #59: Qav configuration

RikusW, 2020-05-25:
Hi, I'm currently investigating Qav on the SJA1105TEL and got it to work to some extent I tested it using iperf3. When I set credit_hi/lo to 0x3fffffff and both send and idle slopes to 10000 the bandwidth was 50% (57MByte/s) as expected and increasing the send slope decreased the bandwidth as expected. However setting the send slope to 9000 suddenly jumps to 72MByte/s. Keeping the send slope at 10000 and increasing the idle slope instead seems to give better results above 50% not sure why. Is there any documentation available on how the Qav hardware works ? I did find sja1105_cbs_commit but it is not used anywhere.

vladimiroltean, 2020-05-27:
Hi Rikus,
Surprised to hear back from you!
A lot has been happening with the sja1105 recently, most importantly it has received a new kernel driver which is now included in OpenIL master branch.
Adding support for the Credit-Based Shaper was one of my last major items on the to-do list for this switch, and now that you asked about it I finally took some time to do it.
Not sure how willing you are to switch to a kernel-based implementation at this point (although there would be a lot of reasons to - it supports 802.1AS using unpatched linuxptp, 802.1Qbv configuration using tc-taprio, 802.1Qci configuration using tc-gate, port mirroring, trapping/dropping/redirecting by DMAC, port, traffic class and flow policers), but at least you can take a look at the branch I set up: https://github.com/vladimiroltean/ls1021atsn-linux/blob/feature/sja1105-cbs/drivers/net/dsa/sja1105/sja1105_main.c#L1681

```diff
net: dsa: sja1105: offload the Credit-Based Shaper qdisc

SJA1105, being AVB/TSN switches, provide hardware assist for the
Credit-Based Shaper as described in the IEEE 8021Q-2018 document.

First generation has 10 shapers, freely assignable to any of the 4
external ports and 8 traffic classes, and second generation has 16
shapers.

We also need to provide a dummy implementation of mqprio qdisc offload,
since this seems to be necessary for shaping any traffic class other
than zero.

The Credit-Based Shaper tables are accessed through the dynamic
reconfiguration interface, so we have to restore them manually after a
switch reset. The tables are backed up by the static config only on
P/Q/R/S, and we don't want to add custom code only for that family,
since the procedure that is in place now works for both.

Tested with the following commands:

data_rate_kbps=67000
port_transmit_rate_kbps=1000000
idleslope=$data_rate_kbps
sendslope=$(($idleslope - $port_transmit_rate_kbps))
locredit=$((-0x80000000))
hicredit=$((0x7fffffff))
tc qdisc add dev swp2 root handle 1: mqprio hw 0 num_tc 8 \
        map 0 1 2 3 4 5 6 7 \
        queues 1@0 1@1 1@2 1@3 1@4 1@5 1@6 1@7
tc qdisc replace dev swp2 parent 1:1 cbs \
        idleslope $idleslope \
        sendslope $sendslope \
        hicredit $hicredit \
        locredit $locredit \
        offload 1

```

I tested it using:
```
# Sender
iperf3 -c 192.168.1.3 -u -b 0                                                                                                                                                             
Connecting to host 192.168.1.3, port 5201
[  5] local 192.168.1.5 port 33269 connected to 192.168.1.3 port 5201
[ ID] Interval           Transfer     Bitrate         Total Datagrams
[  5]   0.00-1.00   sec  37.0 MBytes   310 Mbits/sec  26770
[  5]   1.00-2.00   sec  37.2 MBytes   312 Mbits/sec  26950
[  5]   2.00-3.00   sec  45.3 MBytes   380 Mbits/sec  32810
[  5]   3.00-4.00   sec  47.6 MBytes   400 Mbits/sec  34500
[  5]   4.00-5.00   sec  47.6 MBytes   400 Mbits/sec  34500
[  5]   5.00-6.00   sec  47.6 MBytes   399 Mbits/sec  34470
[  5]   6.00-7.00   sec  47.6 MBytes   400 Mbits/sec  34490
[  5]   7.00-8.00   sec  47.6 MBytes   399 Mbits/sec  34490
[  5]   8.00-9.00   sec  47.6 MBytes   399 Mbits/sec  34490
[  5]   9.00-10.00  sec  47.6 MBytes   400 Mbits/sec  34490
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.00  sec   453 MBytes   380 Mbits/sec  0.000 ms  0/327960 (0%)  sender
[  5]   0.00-10.01  sec  77.5 MBytes  64.9 Mbits/sec  0.024 ms  271836/327957 (83%)  receiver

iperf Done.
# Receiver
iperf3 -s
-----------------------------------------------------------
Server listening on 5201
-----------------------------------------------------------
Accepted connection from 192.168.1.5, port 44254
[  5] local 192.168.1.3 port 5201 connected to 192.168.1.5 port 33269
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-1.00   sec  7.74 MBytes  64.9 Mbits/sec  0.032 ms  20845/26448 (79%)
[  5]   1.00-2.00   sec  7.74 MBytes  64.9 Mbits/sec  0.036 ms  21347/26953 (79%)
[  5]   2.00-3.00   sec  7.74 MBytes  64.9 Mbits/sec  0.020 ms  27109/32715 (83%)
[  5]   3.00-4.00   sec  7.74 MBytes  64.9 Mbits/sec  0.032 ms  28896/34502 (84%)
[  5]   4.00-5.00   sec  7.74 MBytes  64.9 Mbits/sec  0.034 ms  28888/34493 (84%)
[  5]   5.00-6.00   sec  7.74 MBytes  64.9 Mbits/sec  0.028 ms  28875/34481 (84%)
[  5]   6.00-7.00   sec  7.74 MBytes  64.9 Mbits/sec  0.020 ms  28887/34493 (84%)
[  5]   7.00-8.00   sec  7.74 MBytes  64.9 Mbits/sec  0.028 ms  28881/34487 (84%)
[  5]   8.00-9.00   sec  7.74 MBytes  64.9 Mbits/sec  0.026 ms  28884/34490 (84%)
[  5]   9.00-10.00  sec  7.74 MBytes  64.9 Mbits/sec  0.034 ms  28889/34495 (84%)
[  5]  10.00-10.01  sec  91.9 KBytes  63.6 Mbits/sec  0.024 ms  335/400 (84%)
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Jitter    Lost/Total Datagrams
[  5]   0.00-10.01  sec  77.5 MBytes  64.9 Mbits/sec  0.024 ms  271836/327957 (83%)  receiver
-----------------------------------------------------------
```

Notice that I set data_rate_kbps to 67,000 and I got 64.9 Mbps reported by iperf3 (UDP payload), so I'd say it's pretty close. I also tested it across a more varied range of bandwidths and it seems pretty linear.

Hope this helps,
-Vladimir

RikusW, 2020-05-28:
Hi Vladimir

This does help, thank you.
We'll be using a Digi module with a Yocto build, I'm not sure how easy it will be to pull in the latest kernel. I will be testing a SJA1105Q soon so having kernel support will be helpful.

Thanks
Rikus

vladimiroltean, 2020-05-28:
For what it's worth, I'm not working on the sja1105-tool anymore, due to the many design limitations that it has, so it's unlikely that CBS support for it is going to happen.

RikusW, 2020-05-28:
I do have my own SJA register access code for our gPTP so could use that if need be, sja1105-tool also allow direct register access so it could be shell scripted as well. I just wasn't sure exactly how to use the CBS registers.

vladimiroltean, 2020-05-28:
If you plan to migrate to the SJA1105Q and you do raw register access I suppose all of those scripts need to be updated anyway.