nxp-archive/openil_sja1105-tool #60: SJA1105Q only allows broadcast frames through

RikusW, 2020-11-18:
I used the same xml file that works on SJA1105T and just changed the id, the link goes up and ARP broadcasts pass through but the ARP reply does not so ping fails. Any ideas why ?

vladimiroltean, 2020-11-18:
When I started supporting the sja1105q was about the same time that I started working on the kernel driver. So the sja1105q is minimally supported by the sja1105-tool, I'm not even sure if everything is in there.
Would you please switch to the kernel driver? If not, at least you could take a look at how the kernel driver sets up things, so you could replicate that in the xml. I think there are some differences in the L2 Lookup Parameters
https://github.com/torvalds/linux/blob/master/drivers/net/dsa/sja1105/sja1105_main.c#L237

RikusW, 2020-11-18:
Thank you, I will have a look at that.

RikusW, 2020-11-30:
I changed tag_port = 0 on vlan 0 and now it works.

closed 2020-11-30 by RikusW