nxp-archive/openil_sja1105-tool #61: Standalone cross compilation toolchain

Waner98, 2021-02-15:
Hi @vladimiroltean ,

Just a question concerning your example of standalone toolchain. I am actually working on LS1021ATSN and I need to cross compile some libraries. For this I would prefer to have some static libraries, and so having a standalone toolchain too for cross compilation. But with your methods, CMake is not compatible. Do you have any clean solution to pass from buildroot to a standalone toolchain ?

vladimiroltean, 2021-02-15:
> But with your methods, CMake is not compatible.

Have you tried setting
```
export CMAKE_PREFIX_PATH="${SYSROOT_PATH}/usr"
```

> Do you have any clean solution to pass from buildroot to a standalone toolchain ?

I don't understand what it means to "pass from buildroot to a standalone toolchain"/

Waner98, 2021-02-16:
You were right, this env variable was the issue. Thanx for your quick answer !

closed 2021-02-16 by Waner98